blueman-project/blueman #1831: failed to apply network setting

Anas-jaf, 2022-08-19:
when i enable NAP i get this error on the startup 
[image]

blueman: 2.3.1-1
BlueZ: 5.64-2+kali1
Distribution: kali linux raspberry pi 4
Desktop environment: xfce4

cschramm, 2022-08-19:
Assuming 2.3.1-1 is the respective Debian package, it ships the following polkit rules: https://salsa.debian.org/debian/blueman/-/blob/debian/2.3.1-1/debian/blueman.rules

That means, you need to be in the `sudo` or `netdev` groups.

Anas-jaf, 2022-08-20:
i added my user to `netdev` and `sudo` groups with `usermod -aG sudo kali`  and `usermod -aG netdev kali` and  `gpasswd -a kali netdev ` and still having the same issue 

I put my user in visudo 
[image]

also I checked using id the groups 
[image]

thanks in advance

cschramm, 2022-08-21:
Did you restart blueman-applet with the new groups? Maybe try if `pkcheck --action-id org.blueman.network.setup --process $(pgrep blueman-applet)` gives you anything useful.

Anas-jaf, 2022-08-21:
i tried `pkcheck --action-id org.blueman.network.setup --process $(pgrep blueman-applet)` but it returnes with `not authorized`

Anas-jaf, 2022-08-22:
i didn't find how to authorize blueman policy kit so I found this article that discript how to create pan with bluez-tools with the help of this article [article](https://thesmashy.medium.com/raspberry-pi-zero-w-wifi-hacking-gadget-63e3fa1c3c8d) until I find how to fix this issue I will use this method

cschramm, 2022-08-22:
I would have expected a diagnostic message from `pkcheck`. Maybe try with `--allow-user-interaction`.

As I already mentioned if 2.3.1-1 is the respective Debian package, authorization rules are set up as https://salsa.debian.org/debian/blueman/-/blob/debian/2.3.1-1/debian/blueman.rules installed at `/usr/share/polkit-1/rules.d/blueman.rules`.

Just to make sure: This is not a long-running system and you did restart at some point?

Anas-jaf, 2022-08-24:
i'm using raspberrypi kali linux which is based on debian

closed 2022-11-01 by github-actions[bot]