blueman-project/blueman #1833: Bluetooth issue with my mouse after resume from sleep

iant321, 2022-08-20:
blueman: 2.3.2
BlueZ: 5.65
Distribution: Fedora 36
Desktop environment: KDE
Kernel: 5.18.17

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I have a bluetooth mouse Logitech M720 and a Lenovo T460 laptop. Here is the scenario to reproduce my issue. Initially, the laptop is in the sleep state with the lid closed. When I open the lid, I see the kde lock screen. If I begin to move my external mouse immediately after this (as I have to position my mouse pointer in the password field), the mouse pointer does not react, and after unlocking with my trackpad/keyboard I see that bluetooth is off. The blueman icon in the system tray is present, with the red cross over its image. I can click on it and switch bluetooth on, after that my external mouse is connected and works ok. 

However, if, while on the lock screen, I wait for 4-5 seconds, and only after that begin to move the external mouse, it gets connected, the mouse pointer moves and everything is ok.

I have AutoEnable=true in /etc/bluetooth/main.conf. Also, in the [Policy] section ResumeDelay=2. This apparently explains why my mouse is connected on the lock screen if I wait for a few seconds. Before that, bluetooth is off. Comments in main.conf explain that the delay is needed for better coexistence with wifi. I tried to set ResumeDelay=0 but got a lot of messages like "too many simultaneous events", so I returned it to the default value 2.

What I do not understand is why, if I move my external mouse  before bluetooth is switched on, the bluetooth locks in the "off" state. I would expect that my mouse would be irresponsive for a couple of seconds and then would connect after bluetooth switched on.

Thanks for reading this.

cschramm, 2022-08-21:
Sounds super weird, yes.

`AutoEnable` shouldn't really be involved, I think. If the adapter is not powered your mouse could not wake up the system, right? It seem like Bluetooth actually gets locked or unpowered when the system gets woken up by the mouse.

Did you check and compare the bluetoothd logs (and `rfkill` messages from the kernel) for both cases? bluetoothd debugging messages might be helpful as well.

iant321, 2022-08-21:
cschramm, thanks for your answer. Here are the relevant lines from /var/log/messages for the two cases. I try to keep the number of lines from /var/log/messages as small as I can, if I missed something, please let me know.

**When I move my mouse immediately (bluetooth stays off):**

Aug 21 12:12:21 apollo systemd-logind[721]: Operation 'sleep' finished.
...
Aug 21 12:12:21 apollo blueman-applet[1511]: blueman-applet 12.12.21 WARNING  DiscvManager:119 update_menuitems: warning: Adapter is None
...
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Waiting for firmware download to complete
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Firmware loaded in 1634844 usecs
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Waiting for device to boot
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Device booted in 11748 usecs
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Malformed MSFT vendor event: 0x02
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-11-5.ddc
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Applying Intel DDC parameters completed
Aug 21 12:12:22 apollo kernel: Bluetooth: hci0: Firmware revision 0.0 build 14 week 44 2021
Aug 21 12:12:22 apollo bluetoothd[48786]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
Aug 21 12:12:22 apollo bluetoothd[48786]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
Aug 21 12:12:23 apollo bluetoothd[48786]: Endpoint registered: sender=:1.41 path=/MediaEndpoint/A2DPSource/ldac
...
Aug 21 12:12:23 apollo bluetoothd[48786]: src/adv_monitor.c:btd_adv_monitor_power
_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/deviceinfo/deviceinfo.c:read_p
npid_cb() Error reading PNP_ID value: Request attribute has encountered an unlike
ly error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:info_read_cb()
 HID Information read failed: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_read_cb
() Error reading Report value: Request attribute has encountered an unlikely erro
r
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_referen
ce_cb() Read Report Reference descriptor failed: Request attribute has encountere
d an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_read_cb
() Error reading Report value: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo bluetoothd[48786]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Aug 21 12:12:23 apollo blueman-applet[1511]: blueman-applet 12.12.23 WARNING  PowerManager:203 on_adapter_property_changed: adapter powered on while in off state, turning bluetooth on
Aug 21 12:12:23 apollo systemd[1]: Starting blueman-mechanism.service - Bluetooth management mechanism...
Aug 21 12:12:23 apollo systemd[1]: Started blueman-mechanism.service - Bluetooth management mechanism.

**When I do not move my external mouse on the lock screen and move it after unlocking.**

Aug 21 12:27:27 apollo systemd-logind[721]: Operation 'sleep' finished.
...
Aug 21 12:27:27 apollo blueman-applet[1511]: blueman-applet 12.27.27 WARNING DiscvManager:119 update_menuitems: warning: Adapter is None
...
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Waiting for firmware download to complete
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Firmware loaded in 1650430 usecs
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Waiting for device to boot
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Malformed MSFT vendor event: 0x02
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Device booted in 11783 usecs
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-11-5.ddc
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Applying Intel DDC parameters completed
Aug 21 12:27:28 apollo kernel: Bluetooth: hci0: Firmware revision 0.0 build 14 week 44 2021
Aug 21 12:27:28 apollo bluetoothd[48786]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
[... cut ...]
Aug 21 12:27:38 apollo kernel: input: M720 Triathlon Keyboard as /devices/virtual/misc/uhid/0005:046D:B015.0010/input/input76
Aug 21 12:27:38 apollo kernel: input: M720 Triathlon Mouse as /devices/virtual/misc/uhid/0005:046D:B015.0010/input/input77
ug 21 12:27:38 apollo kernel: hid-generic 0005:046D:B015.0010: input,hidraw0: BLUETOOTH HID v0.13 Keyboard [M720 Triathlon] on e4:a7:a0:8d:fc:79
Aug 21 12:27:38 apollo systemd-logind[721]: Watching system buttons on /dev/input/event5 (M720 Triathlon Keyboard)

I have no idea why "Keyboard" appears in the description of my mouse.

cschramm, 2022-08-23:
I'd clearly blame it on all those profiles/input/hog-lib.c errors. The BlueZ project should be much more helpful with this, then.

I have to add that I do no really get how the wake works as it happens before hci0 gets (re-)initialized and in my mind the input device should not exist before that. There might be some "light mode" the adapter and input device are in to wake up the system.

iant321, 2022-08-23:
Thank you for your comment and for narrowing the possible cause for th problem. I will now try to look at the bluez forum.

closed 2022-10-31 by github-actions[bot]