blueman-project/blueman #1836: Samsung Soundbar HW-H450 paired, trusted but cannot connect with errror br-connection-page-timeout

vgazioglu, 2022-08-21:
blueman: applet 2.3.
BlueZ: bluetoothd  --version
5.64

Distribution: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy Kernel: 5.15.0-46-generic x86_64 bits: 64 compiler: gcc v: 11.2.0 
Desktop environment: Desktop: Cinnamon 5.4.11
    tk: GTK 3.24.33 wm: Mutter dm: LightDM 
* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After upgrading to Linux Mint 21, I cannot use my Samsung soundbar HW-H450 anymore even though i can pair and trust it and see services icons related on the applet. 

[image]

  Device-1: Intel Bluetooth wireless interface type: USB driver: btusb v: 0.8 bus-ID: 1-1.3:3
    chip-ID: 8087:0a2a
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.2
    sub-v: 1000

syslog
Aug 22 01:34:54 vegae7250 systemd[2125]: Started Bluetooth Manager.
Aug 22 01:35:05 vegae7250 systemd[2125]: Reached target Bluetooth.
Aug 22 01:36:14 vegae7250 bluetoothd[894]: src/device.c:search_cb() BC:14:85:D3:XX:XX: error updating services: Input/output error (5)
Aug 22 01:36:14 vegae7250 blueman-manager[8736]: blueman-manager 01.36.14 WARNING  ManagerDeviceMenu:141 fail      : fail g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusFailedError:  br-connection-create-socket
Aug 22 01:36:14 vegae7250 blueman-manager[8736]:  (0)

cschramm, 2022-08-22:
`bluetoothd[894]: src/device.c:search_cb() BC:14:85:D3:XX:XX: error updating services: Input/output error (5)` is currently the root cause here. The error code comes from the kernel side.

We had similar issues in #158. Most of them seem to boil down to some pulseaudio misbehavior caused by `module-bluetooth-discover` not being loaded at some point or something. I'd guess that's a thing of a past but maybe not. Did you check logs from pulseaudio?

vgazioglu, 2022-08-22:
Doyou mean the below?

N: [pulseaudio] bluez5-util.c: Could not find org.bluez.BatteryProviderManager1.RegisterBatteryProvider(), is bluetoothd started with experimental features enabled (-E flag)?
E: [pulseaudio] backend-native.c: org.bluez.ProfileManager1.RegisterProfile() failed: org.bluez.Error.NotPermitted: UUID already registered
E: [pulseaudio] backend-native.c: org.bluez.ProfileManager1.RegisterProfile() failed: org.bluez.Error.NotPermitted: UUID already registered
I: [pulseaudio] module-suspend-on-idle.c: Source alsa_input.pci-0000_00_1b.0.analog-stereo idle for too long, suspending ...
I: [alsa-source-ALC3235 Analog] alsa-source.c: Device suspended...
I: [pulseaudio] module-suspend-on-idle.c: Sink alsa_output.pci-0000_00_1b.0.analog-stereo idle for too long, suspending ...
I: [alsa-sink-ALC3235 Analog] alsa-sink.c: Device suspended...
I: [pulseaudio] module-card-restore.c: Synced.
I: [pulseaudio] module-device-restore.c: Synced.
I: [pulseaudio] core.c: We are idle, quitting...
I: [pulseaudio] main.c: Daemon shutdown initiated.

cschramm, 2022-08-26:
Basically yes. What's interesting is the moment of when you're trying to connect to the device (e.g. ~Aug 22 01:36:14).

I'm wondering about those error messages about already registered profiles. Could it be that you started multiple pulseaudio instances? :thinking:

closed 2022-11-02 by github-actions[bot]