blueman-project/blueman #1838: Disabling ConnectionNotifier plugin does not take effect

ba32107, 2022-08-28:
blueman: `2.2.4` (shown on the About dialog of Bluetooth Manager)
BlueZ: not sure how to get this
Distribution: Ubuntu 22.04.1
Desktop environment: Gnome

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi,

Opening this issue after being recommended in https://github.com/blueman-project/blueman/issues/1556. Copy pasting the description from there:

> I have tried disabling the ConnectionNotifier plugin, and it works until I restart my computer. Upon restart, I again see the notifications. When I open blueman, I go into plugins, I see that the ConnectionNotifier is disabled, but the app also crashes. After that happens, the notifications are not shown as expected.
> So, it's as if I need to open blueman and the Plugins dialog for blueman to actually notice that this plugin should be disabled.

As recommended, I have ran `blueman-applet --loglevel debug`. This is the output:
https://pastebin.com/rb6ym4mY

Whenever I start Bluetooth Manager and reproduce the crash, the output of `blueman-applet` does not change.

I have also ran `blueman-manager --loglevel debug`. Output:
https://pastebin.com/kkhFxriU

After reproducing the crash (opening the `Plugins` dialog:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 324, in __async_result_handler
    result_callback(obj, self._unpack_result(ret), real_user_data)
TypeError: ManagerMenu._on_plugin_dialog_activate.<locals>.cb() takes 0 positional arguments but 3 were given

```

Thanks!

cschramm, 2022-08-28:
The error on opening the plugins dialog from the manager menu is fixed in 2.3 (#1733). You can still open the dialog from the tray icon's menu.

From the log, the `ConnectionNotifier` plugin clearly does not get loaded. What you're saying is that you still get `Connected` and ` Disconnected` notifications? I'm wondering if it's really blueman's notifications. gnome-bluetooth is not in place?

ba32107, 2022-08-29:
> The error on opening the plugins dialog from the manager menu is fixed in 2.3 

Might be a dumb question, but how do I update? I don't have any pending updates from `apt update` and I don't recall manually installing `bluetooth-manager`.

> What you're saying is that you still get Connected and Disconnected notifications

Correct, although I am not sure if the notification I see is coming from blueman. It looks like this (it's my mouse):
[image]

> gnome-bluetooth is not in place?

Don't think so - I can't find on the command line and I never installed it explicitly (I think)

cschramm, 2022-08-29:
That's probably from blueman's `Battery` plugin.

Software updates are not part of Ubuntu's release concept, so that error will stay in 22.04 forever. However, you can e.g. just grab the 2.3.2 deb from kinetic: https://packages.ubuntu.com/kinetic/amd64/blueman/download.

ba32107, 2022-08-29:
You're right! Disabling the battery plugin did the trick. Thanks! 

> Software updates are not part of Ubuntu's release concept

I don't understand what you mean by this. To be honest, I don't remember how blueman is installed (don't remember if I added the PPA manually or if it was part of core Ubuntu), but as long as you publish 2.3.2 into the corresponding `apt` repo, I should get the new version automatically. 

Are you saying that 2.3.2 is only available for kinetic, not Jammy Jellyfish?

(I'm going to close the issue btw because my problem is solved, thanks)

closed 2022-08-29 by ba32107

cschramm, 2022-08-29:
> Are you saying that 2.3.2 is only available for kinetic, not Jammy Jellyfish?

Exactly. Ubuntu releases like jammy do not get any new software versions except security updates (into their apt repositories).