blueman-project/blueman #1839: ELM 237 Serial Port

DIGIFAULT, 2022-08-29:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2022-08-29:
We really don't need any info at all. At least do the minimal effort of providing what we ask in the template.

closed 2022-08-29 by infirit

DIGIFAULT, 2022-08-30:
DDT4ALL With python3 - Serial Port Error (unrecognized) on fake Chinese ELM 237 and Bluetooth USB ASUS BT500, not recognized. Toshiba, 64 bites, Linux Mint 20...
 ---------------------------------------------------------------
class SerialBase(io.RawIOBase):
    """\
    Serial port base class. Provides __init__ function and properties to
    get/set port settings.
---------------------------------------------------------------