blueman-project/blueman #1842: Feature Request -- Remember Bluetooth On/Off Setting

mfreeman72, 2022-09-01:
blueman: 2.3.2
BlueZ: 5.64
Distribution: Linux Mint 21
Desktop environment: Cinnamon

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Would it be possible to have Blueman remember its Bluetooth on/off setting across a reboot? Currently, if I turn Bluetooth off in the menu, the next time I reboot, Bluetooth is on again. I'd like to have the option of having this setting stay the way I set it, either on or off, even after rebooting, until I tell it to change.

cschramm, 2022-09-02:
There are multiple levels of on / off state. Most probably this is about powering the adapter (running `bluetoothctl power on` should do the trick then).

BlueZ has you covered with its `AutoEnable` setting then. We also do have a legacy option in blueman-applet's `PowerManager` plugin. However, blueman's option is on by default and Ubuntu's `bluez` packages ship `AutoEnable=true` (side note: This is the upstream default, starting with 5.65), so everything should work as expected already. We did receive a couple of reports from Linux Mint users that it does not, but we never heard of feedback why.

mfreeman72, 2022-09-02:
I've experimented with all three of the options you mentioned:

-- `bluetoothctl power [on/off]` seems to just turn bluetooth directly on/off, and doesn't seem to affect its state at reboot time, so this doesn't seem to affect my issue one way or another.

-- I've tried all four permutations of settings with Blueman's PowerManager plugin and Bluez's AutoEnable setting:
        1. Plugin on and AutoEnable on = Bluetooth always on at startup, even if it was off at shutdown.
        2. Plugin off and AutoEnable on = Bluetooth always on at startup, even if it was off at shutdown.
        3. Plugin on and AutoEnable off = Bluetooth always on at startup, even if it was off at shutdown.
        4. Plugin off and AutoEnable off = Bluetooth always off at startup, even if it was on at shutdown.
    None of those are what I want to happen.

What I'm looking for is: If bluetooth is off when I shutdown/reboot the computer, it should be off at startup. And if bluetooth is on when I shutdown/reboot the computer, it should be on at startup. I can't seem to get this to happen, no matter what any of the settings are set to.

cschramm, 2022-09-03:
What you're looking for is a killswitch, not the power state of the adapter. Killswitches are usually persistent and the system typically has one for Bluetooth. You can check the states with `rfkill`.

blueman does override soft killswitches in its default configuration (as the whole point of running blueman is using Bluetooth). To avoid that, beside not starting blueman if you're not actually going to use Bluetooth, you can either turn off the `PowerManager` setting so that it does not try things on its own or disable the `KillSwitch` plugin so that state changes do not get propagated to killswitches.

I'm surprised that permutation 2 does not have the desired effect as I'd expect that you have a killswitch, that it's persistent and that nothing overrides the killswitch if the blueman option is disabled. That requires that the killswitch got set before shutdown of course, but if you disable Bluetooth in blueman it takes care of it.

closed 2022-11-10 by github-actions[bot]