blueman-project/blueman #1844: Gtk CRITICAL Error while trying to dockerise blueman

thesuhas, 2022-09-05:
blueman: 2.02
BlueZ: 5.53
Distribution: Ubuntu 20.04
Desktop environment: Ubuntu 20.04 in a Docker Container on Pop OS 22.04

Hi, 
I am trying to dockerise an older version of blueman for a personal project of mine.
I've installed all the required dependencies. When trying to run `blueman-manager` I am getting a Gtk-CRITICAL Error and I am unable to fix it.
Attaching a screenshot for your reference.

[image]

infirit, 2022-09-06:
Can you run it through gdb and get a backtrace. That should tell you where
it faults.

cschramm, 2022-09-07:
GTK cannot connect to the display server. I guess the errors are more or less expected under some circumstances then.

With just `apt update; apt install blueman; /etc/init.d/dbus start; blueman-manager` I get

```
Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 44, in <module>
    manager = Blueman()
  File "/usr/lib/python3/dist-packages/blueman/main/Manager.py", line 29, in __init__
    super().__init__(title=_("Bluetooth Devices"))
  File "/usr/lib/python3/dist-packages/gi/overrides/Gtk.py", line 515, in __init__
    raise RuntimeError(
RuntimeError: Gtk couldn't be initialized. Use Gtk.init_check() if you want to handle this case.
```

Might be something else in older versions (I don't get what you mean with `2.02`).

thesuhas, 2022-09-08:
@cschramm 
I meant the 2.02 release of blueman

cschramm, 2022-09-08:
There is no such release. Possibly 2.0.2 or 2.2.

thesuhas, 2022-09-08:
@cschramm 
My bad, I meant 2.0.2

cschramm, 2022-09-08:
I can confirm that error in that ancient version. In 2.0 the first call to `Gtk.Builder.add_from_file` causes a crash when there's no connection to the display server.

Not sure about your expectations here. blueman-manager is not expected to work without a display server.

closed 2022-11-16 by github-actions[bot]