blueman-project/blueman #1850: Missed file in 2.3.2 asset

kuraga, 2022-09-09:
[blueman-2.3.2.tar.xz](https://github.com/blueman-project/blueman/releases/download/2.3.2/blueman-2.3.2.tar.xz) asset (and, maybe, others) doesn't contain `blueman/plugins/applet/PulseAudioProfile.py` file (and, maybe, others).

infirit, 2022-09-09:
Because it's in plugins/manager

kuraga, 2022-09-09:
@infirit is it another?

kuraga, 2022-09-09:
Ok but such plugin is missed in Gentoo package's UI.
And file is missed, too.
Coincidence?

infirit, 2022-09-09:
You will need to enable the pulseaudio use flag for it to be installed https://packages.gentoo.org/packages/net-wireless/blueman

infirit, 2022-09-09:
The applet plugin was removed in 62548f74262a9125afca75a88b579516190d8ceb but even that never used `plugins/applet/PulseAudioProfile.py` :thinking: . @cschramm do you have any idea why we have this file there ?

@kuraga we only have the manager plugin now. Enable the use flag and it will be in the device right click menu.

cschramm, 2022-09-10:
Yeah, well, looks like that was missed in #1617. :grimacing:

infirit, 2022-09-10:
I see the items being made but the submenu is empty. I'm really not familiar enough with the whole statusnotifier to comment on why. Below diff adds the relevant bits to automake and meson.

```diff
diff --git a/blueman/plugins/applet/Makefile.am b/blueman/plugins/applet/Makefile.am
index 6f174c85..833c4808 100644
--- a/blueman/plugins/applet/Makefile.am
+++ b/blueman/plugins/applet/Makefile.am
@@ -27,6 +27,9 @@ blueman_PYTHON = \
     StatusNotifierItem.py \
     TransferService.py
 
+if HAVE_PULSEAUDIO
+blueman_PYTHON += PulseAudioProfile.py
+endif
 
 CLEANFILES =           \
        $(BUILT_SOURCES)
diff --git a/meson.build b/meson.build
index a552180d..d7679a3d 100644
--- a/meson.build
+++ b/meson.build
@@ -332,6 +332,7 @@ blueman_exclude_files = ['Constants.py.in'] + blueman_makefiles
 pulseaudio = get_option('pulseaudio')
 if not pulseaudio
     blueman_exclude_files += 'plugins/manager/PulseAudioProfile.py'
+    blueman_exclude_files += 'plugins/applet/PulseAudioProfile.py'
 endif
 
 install_subdir(
```

```
_generate_profiles_menu: [{'text': 'Uit', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35cca0>, 'tooltip': ''}, {'text': 'High Fidelity Playback (A2DP Sink)', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35ce50>, 'tooltip': ''}, {'text': 'Headset Head Unit (HSP/HFP)', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35cee0>, 'tooltip': ''}, {'text': 'High Fidelity Playback (A2DP Sink, codec SBC)', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35cdc0>, 'tooltip': ''}, {'text': '<b>High Fidelity Playback (A2DP Sink, codec SBC-XQ)</b>', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35d000>, 'tooltip': ''}, {'text': 'High Fidelity Playback (A2DP Sink, codec aptX)', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35da20>, 'tooltip': ''}, {'text': 'Headset Head Unit (HSP/HFP, codec CVSD)', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35dab0>, 'tooltip': ''}, {'text': 'Headset Head Unit (HSP/HFP, codec mSBC)', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'sensitive': True, 'callback': <function AudioProfiles.add_device_profile_menu.<locals>._activate_profile_wrapper.<locals>._wrapper at 0x7f744d35db40>, 'tooltip': ''}]
```

cschramm, 2022-09-11:
:thinking: They show up on my side and the items do not differ from the pre-#1799 recent connections submenu.

infirit, 2022-09-14:
I'll check xfce as it's not looking good in plasma.

infirit, 2022-09-14:
It's fine on XFCE but broken under plasma. I'll see if I can find a commit where it worked. 
[image]

infirit, 2022-09-14:
It's all submenu's that are broken, recentconns is also broken on plasma.

infirit, 2022-09-14:
The moment `AppIndicator` 5ff2a8f2d38ed00aac37f354c5a7a6bcd705e77e is removed and `StatusNotifier` is now generating the menu it breaks.

I never noticed as I don't use RecentConns.

closed 2022-09-15 by cschramm

cschramm, 2022-09-17:
They will be part of a 2.4 release and possibly 2.3.3.

kuraga, 2022-09-17:
@cschramm , I asked exactly about GitHub assets (`.tar.gz`, `.tar.xz`). Because I don't know how are they constructed and don't see if files are added to them in #1855. Thanks.

cschramm, 2022-09-19:
Not sure what you mean with GitHub assets (if not the release tarballs) and whether you're asking for additional information.

kuraga, 2022-09-19:
I mean: [`blueman/plugins/applet/PulseAudioProfile.py`](https://github.com/blueman-project/blueman/blob/2.3.2/blueman/plugins/applet/PulseAudioProfile.py) has to exist not only after `make && make install` but also in [release tarrballs (`.tar.gz`, `.tar.xz`)](https://github.com/blueman-project/blueman/releases/tag/2.3.2).

For now, it doesn't exist in tarballs. And I don't know if #1855 fixes this.

cschramm, 2022-09-20:
It will get added neither to the archives nor the build configurations of 2.3.2 retroactively.

kuraga, 2022-09-20:
> It will get added neither to the archives nor the build configurations of 2.3.2 retroactively.

Talked about next release, of course.

Understand. Thanks.

PR #1855 (2022-09-15): Actually ship PulseAudioProfile applet plugin
Closes #1850
commit 6a2af419 (2022-09-15): Actually ship PulseAudioProfile applet plugin

Closes #1850