blueman-project/blueman #1852: blueman stuck on searching and not showing up any devices

princessclocky, 2022-09-11:
blueman:2.3.1
BlueZ:
Distribution: 
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

princessclocky, 2022-09-11:
ok so I don`t know if i did  somthing wrong but blueman does not recognize any bluetooth devices, i click on search and the loading bar moves but nothing else happens, I TRIED EVERYTHING, even restored the system, but nothing works can someone help? I`m using linux mint 21 on a hp elitebook 8460p
[image]

infirit, 2022-09-12:
Have a look through https://github.com/blueman-project/blueman/wiki/Troubleshooting.

fran6120, 2022-09-12:
Same problem here.

I am using Linux Mint 21 too.

[image]

In my case running `lsusb && refkill` show that usb bluetooth adapter is attatched and ready.

```
~$ lsusb && rfkill
Bus 001 Device 003: ID 0bda:5776 Realtek Semiconductor Corp. HP Truevision HD integrated webcam
Bus 001 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 046d:c019 Logitech, Inc. Optical Tilt Wheel Mouse
Bus 002 Device 002: ID 0bda:a725 Realtek Semiconductor Corp. Bluetooth 5.1 Radio
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
ID TYPE      DEVICE      SOFT      HARD
 0 bluetooth hci0   unblocked unblocked
 1 wlan      phy0   unblocked unblocked
```

So runnig first `systemctl restart bluetooth.service` then `bluetoothctl` for some reason aparently works...

[image]

If i try to pair seem works:

```
~$ bluetoothctl
Agent registered
[CHG] Controller 00:E0:42:B7:09:03 Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:E0:42:B7:09:03 Discovering: yes
[NEW] Device D8:6C:02:4D:C3:42 realme Buds Air 3
[NEW] Device B4:20:5B:30:F3:52 OPPO Find X3 Lite 5G
[bluetooth]# pair D8:6C:02:4D:C3:42
Attempting to pair with D8:6C:02:4D:C3:42
[CHG] Device D8:6C:02:4D:C3:42 Connected: yes
[CHG] Device D8:6C:02:4D:C3:42 Modalias: bluetooth:v0094p0123d1000
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 00000000-0000-0000-0099-aabbccddeeff
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 0000079a-d102-11e1-9b23-00025b00a5a5
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device D8:6C:02:4D:C3:42 UUIDs: df21fe2c-2515-4fdb-8886-f12c4d67927c
[CHG] Device D8:6C:02:4D:C3:42 ServicesResolved: yes
[CHG] Device D8:6C:02:4D:C3:42 Paired: yes
Pairing successful
[CHG] Device D8:6C:02:4D:C3:42 ServicesResolved: no
[CHG] Device D8:6C:02:4D:C3:42 Connected: no
[CHG] Device D8:6C:02:4D:C3:42 Trusted: yes
[CHG] Device D8:6C:02:4D:C3:42 Connected: yes
[CHG] Device D8:6C:02:4D:C3:42 ServicesResolved: yes
[realme Buds Air 3]# 

```
[image]

But unfortunately, pulseaudio daemon stop and gets into restarting-stopping loop.

I can confirm that this adapter works on Linux Mint 20.3 and Windows.

cschramm, 2022-09-12:
@fran6120: It looks like you're reporting an issue with pulseaudio in Linux Mint 21. This has nothing to do with @princessclocky's problem that no devices show up when searching and blueman is not the correct project to address (either Linux Mint or pulseaudio).

closed 2022-11-21 by github-actions[bot]