blueman-project/blueman #1853: blueman-manager: ValueError: Namespace Gtk not available

rucete, 2022-09-12:
blueman: 2.3.2-1
BlueZ: 5.64-2
Distribution:  Debian 5.18.5-1kali6 x86_64
Desktop environment: Xfce

Hi,

When starting blueman-manager I get the following message:

> $ blueman-manager
> 
> Traceback (most recent call last):
>   File "/usr/bin/blueman-manager", line 16, in <module>
>     from blueman.main.Manager import Blueman
>   File "/usr/lib/python3/dist-packages/blueman/main/Manager.py", line 10, in <module>
>     from blueman.Functions import *
>   File "/usr/lib/python3/dist-packages/blueman/Functions.py", line 44, in <module>
>     gi.require_version("Gtk", "3.0")
>   File "/usr/lib/python3/dist-packages/gi/\_\_init\_\_.py", line 126, in require_version
>     raise ValueError('Namespace %s not available' % namespace)
> ValueError: Namespace Gtk not available

I can't debug because `blueman-applet` also crashes this way. I've checked that I have installed the required dependencies, especially `python3-gi` and `gir1.2-gtk-3.0` but I'm taking suggestions.

This issue persists in any other python program installed in my computer that uses Gtk but I've searched a lot for a system-wide solution and found nothing. Currently, I'm trying to determine how does `gi.Repository.get_default().enumerate_versions(Gtk)` works (this is line 124 in `/usr/lib/python3/dist-packages/gi/__init__.py`) and determine if my python installation is not searching correctly for Gtk but the documentation is very scarce.

Any help is appreciated.

* [Yes] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2022-09-12:
If this happens with any python script then my suggestion is to ask in your distro's support channels. I think you're on Kali so see https://www.kali.org/community/

cschramm, 2022-09-12:
(Something seems really fucked up in your system but in general...)

The gi module doesn't do anything fancy. It just delegates to the default gobject-introspection stuff.

`gir1.2-gtk-3.0` ships `/usr/lib/x86_64-linux-gnu/girepository-1.0/Gtk-3.0.typelib`.

You can check the default path with following C program if you like:

```c
#include <stdio.h>
#include <girepository.h>

int main(void) {
    g_irepository_get_default();
    g_slist_foreach(g_irepository_get_search_path(), (GFunc)puts, NULL);
    return 0;
}
```

Just `gcc $(pkg-config --cflags --libs gobject-introspection-1.0)` it with `libgirepository1.0-dev` installed and run the resulting `a.out`. It should point you to the typelib's path.

rucete, 2022-09-13:
@cschramm Thank you! So I ran the code and found `girepository-1.0` and `gobject-introspection` under `/usr/local/lib/x86_64-linux-gnu` but when I checked inside, there were a lot of files ''missing'', especially the one related with Gtk, so I searched for a file named `Gtk-3.0.typelib` and found another location for girepo and gobject directories, this time under `/usr/lib/x86_64-linux-gnu`. Theese directories contained in fact the libraries that I had been installing to try and solve the problem.

Something is wrong either with my python installation or with my OS, as arp installs in one place and python searches for it in another but for the moment, I just ''hid'' (not deleted, just in case) the old directories and created two symlinks under /usr/local... that point to the two libraries in /usr/lib/... and that solved the problem for now. My bluetooth works again! 

If you have more suggestions for why might that be, drop them here. I'll keep looking in other forums but thank you very much for the help.

cschramm, 2022-09-13:
If it prints only one path, that one is hardcoded into the loaded `libgirepository-1.0.so.1` and that library is thus not the system one but probably something you put into `/usr/local`.

closed 2022-09-13 by cschramm