blueman-project/blueman #1854: Blueman spaming endless notifications of connected/disconnected headset/speaker

Flashwalker, 2022-09-13:
blueman: 2.2.4
BlueZ: 5.64
Distribution: Pop_Os!
Desktop environment: xfce 4.17

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

￼
When my paired bluetooth headset or speaker is turned on and not yet connected to anywhere,
or when it was connected to PC and i disconnect it from Blueman gui,
Blueman starts endles spaming of notifications of connected/disconnected headset/speaker. It Looks like endless retrying to connect. But it doesn't really connect.

When i connect my bluetooth headset/speaker to any device (PC or smartphone),
or turn the headset/speaker off and on again so Blueman autoconnect it,
or turn the headset/speaker off at all
then spam stops.

And even if i manually disconnect a headset/speaker from Blueman gui and then turn the headset/speaker off and on again - Blueman autoconnect it. Earlier Blueman didn't autoconnect it if it was purposely turned off.

How can i opt-out endless notifications spam?

cschramm, 2022-09-13:
The two main reasons for automatically established connections are that either a) you explicitly configured blueman to auto-connect some of the device's services or b) it's the device that initiates the connection and it's marked trusted on your side.

a) Right-click on the device to see if the respective box is checked.
b) Kill blueman-applet to see if the device still auto-connects without it ("trust" is controlled by the underlying BlueZ system, not blueman itself).

(You can turn of connect and disconnect notifications by disabling the `ConnectionNotifier` plugin but the connections themselves do not seem to be your problem.)

Flashwalker, 2022-09-13:
Well, headset/speaker by itself allways trying to autoconnect when it turned on, by design.

In case i uncheck a trusted checkbox, it doesn't help, and give me two results:
either spam/"attempts" of connect/disconnect still exist or it leads to endless prompts for authorisaton/pairing a device.
Thus, either the Blueman or the headset/speaker continues to auto-connect, but if the Blueman is working correctly, the connection would have been successfully established, isn't it?
But during these endless spam/"attempts" of connect/disconnect there is no real connection. I describe below why.

**Let me describe good former behaviour i has before upgrade my system to fresh release and one day added a bluetooth speaker to Blueman's known devices list (previously there was only a bluetooth TWS headset):**  
The headset was paired and marked as trusted. And if i disconnect it from Blueman, there was no endless spam/"attempts" connect/disconnect notifocations.
Also i have a smartphone, and if i connect and disconnect the headset from smartphone, there is no endless attempts too. Headset stays in detection mode until i connect to it from Blueman or from my smartphone.

**And let me more precisely describe the issue:**  
Blueman starts endles spaming of notifications of connected/disconnected headset/speaker right after i purposefully disconnect it in Blueman gui. It Looks like endless retrying to connect, but it doesn't really connect. When it really connects, i can hear a sound notification in headset/speaker, which says: "Connected".
But during endless spam/"attempts" there is no sound notification: "Connected" in headset/speaker, just silence. So it looks like Blueman is trying to connect and instantly right after this it disconnect, without real connection. And it repeats and repeats endless until i connect manually or turn of the headset/speaker or Blueman.

cschramm, 2022-09-14:
> In case i uncheck a trusted checkbox, it doesn't help, and give me two results: either spam/"attempts" of connect/disconnect still exist or it leads to endless prompts for authorisaton/pairing a device (endless if i dismiss pairing notifications).

Please elaborate on the "either or" in that description. :sweat_smile: Authorisation and paring prompts are also different observations.

Service authorisation prompts clearly indicate that a connection is initiated remotely. Automatically initiated pairing surprises me from either side. Ultimately you should be able to see what's happening if you enable debug logging for `bluetoothd`.

Also, please elaborate on the upgrade you mention. Connect and disconnect notifications got added in blueman 2.2, so if you've been on 2.1 it makes a lot of sense that you did not have them, however, the underlying loop was most probably present just as well, assuming the same Linux kernel, BlueZ, audio server etc.

> **And now let me describe the case i kill Blueman applet during endless spam:** Notifications gone, but the headset/speaker or Bluez is still trying to autoconnect, i can say that because i hear the sound notification during endless spam (typical for connection attempts): "Connection...".
> 
> So it looks like this issue is about Bluez. In fact it happened right after i added a bluetooth speaker to Blueman's known devices list besides the TWS headset.
> 
> And i think it's Bluez or bluetoothd issue, because it trying to connect all the time despite i unchecked trusted checkbox from both devices:
> 
> ```
> ...
> /var/log/syslog:158695:Sep 13 22:38:58 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158705:Sep 13 22:39:03 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158715:Sep 13 22:39:14 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158716:Sep 13 22:39:19 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158725:Sep 13 22:39:30 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158734:Sep 13 22:39:35 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158743:Sep 13 22:39:46 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158744:Sep 13 22:39:51 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158753:Sep 13 22:40:02 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158762:Sep 13 22:40:07 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158771:Sep 13 22:40:18 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158772:Sep 13 22:40:23 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158781:Sep 13 22:40:34 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158790:Sep 13 22:40:39 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158799:Sep 13 22:40:50 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158808:Sep 13 22:40:55 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> /var/log/syslog:158820:Sep 13 22:41:06 smarttop bluetoothd[1193]: src/device.c:search_cb() F4:4E:FD:03:80:98: error updating services: Host is down (112)
> /var/log/syslog:158822:Sep 13 22:41:11 smarttop bluetoothd[1193]: src/device.c:search_cb() 00:00:AB:CE:47:FE: error updating services: Host is down (112)
> ...
> ```

Wait what? Something on your system tries to connect to two paired, untrusted audio devices that are switched off (and it's not blueman as blueman-applet isn't running)? Oh, nice. :smile:

You could check the bus for method calls to bluetoothd e.g. with `sudo dbus-monitor --system destination=org.bluez,interface=org.bluez.Device1`.

Pop_Os! 22.04 switched its audio server to PipeWire. That might be related somehow, but I really would not expect it to send random connect calls to BlueZ.

I'd say chances are high that this is a known issue in Pop_Os! 22.04.

cschramm, 2022-09-14:
So `:1.59` seems to be your villain in that case (the name will probably differ next time). You can try to find the respective process e.g. in `d-feet`. I'm a little bit surprised to see only one of the two devices from your previous log, though.

Flashwalker, 2022-09-14:
> I'm a little bit surprised to see only one of the two devices from your previous log, though.

I removed the second one to see the difference, but it didn't make a difference

Flashwalker, 2022-09-14:
Thanks for `d-feet` tip!
I found the villain! 😁️

[image]
[image]

I have had **kdeconnect** installed for a long time ago, but I definitely remember the **nuntius** came with last system upgrade.
I just killed the **nuntius** process, and notification spam spam and reconnection attempts are gone. At last!

Tanks for your help 👍️

closed 2022-09-14 by Flashwalker