blueman-project/blueman #1856: Submenus do not work in KDE Plasma

cschramm, 2022-09-15:
It's all submenu's that are broken, recentconns is also broken on plasma.

_Originally posted by @infirit in https://github.com/blueman-project/blueman/issues/1850#issuecomment-1247088923_

kuraga, 2022-09-17:
#1815, #1834, #1845, #1847, #1791, #1803 ?

cschramm, 2022-09-19:
There's a fix in #1859.

@infirit: I'm surprised that you do not have any issues with KDE. While replicating and testing this, I've seen #1845 there (not #1791 as #1803 does not fix it). I've used a Manjaro 21.3.7 live system (Plasma 5.24.6).

infirit, 2022-09-19:
I really don't. Maybe it is a problem but I just don't see it. 

```
Operating System: Arch Linux
KDE Plasma Version: 5.25.5
KDE Frameworks Version: 5.98.0
Qt Version: 5.15.6
Kernel Version: 5.19.8-xanmod1-1 (64-bit)
Graphics Platform: X11
Processors: 12 × AMD Ryzen 5 5600X 6-Core Processor
Memory: 31,3 GiB of RAM
Graphics Processor: AMD Radeon RX 590 Series
```

closed 2022-09-19 by cschramm

cschramm, 2022-09-20:
Hrm. :thinking: I just run manjaro-kde-21.3.7-220816-linux515.iso in QEMU with a forwarded USB Bluetooth adapter, then `sudo pacman -Sy blueman && blueman-applet`, turn Bluetooth off (it's on by default) and end up with this:

[image]

Plasma 5.24.6, Frameworks 5.96.0, Qt 5.15.5. A full system update (Plasma 5.25.5, Frameworks 5.97.0) changes the look a little bit but not the glitch.

Anyway, #1847 fixes it, so we're fine but that fix is of higher importance than expected. Might be worth a 2.3.3.

PR #1859 (2022-09-19): Implement com.canonical.dbusmenu.AboutToShow
KDE Plasma expects this method to be implemented. Otherwise it just does not show submenus.

Closes #1856
commit 817bdbd1 (2022-09-19): Implement com.canonical.dbusmenu.AboutToShow

KDE Plasma expects this method to be implemented. Otherwise it just does not show submenus.

Closes #1856
commit 24cc2b84 (2022-10-06): Implement com.canonical.dbusmenu.AboutToShow

KDE Plasma expects this method to be implemented. Otherwise it just does not show submenus.

Closes #1856