blueman-project/blueman #1861: Some applet menu Idea on main branch

souravdas142, 2022-09-21:
Hi @cschramm and @infirit 

I have some Ideas on applet menu, like below

[image]

1. You can see I have multiple bluetooth audio device connected, but still they are not showing in applet menu to disconnect them.
I want each device should show to applet menu with disconnect option like previous (2.3.2) version has.

2. `Audio profile` for each connected device should have menu

3. Is it possible to show battery indicator with `text percentile`  beside each connected devices in applet menu ? I will be more convenient to show beside them instead opening the window. is not it ? please see the above screenshot for example. 

Thank you.

cschramm, 2022-09-21:
Points 1 and 2 are a bug in main, fixed in #1862.

Regarding point three, there's a suggestion to add the battery percentage to the tray icon tooltip in #1797. While that makes a lot of sense to me, the disconnect menu items seem like the wrong place to me.

Point 4 is possible, yes, but I'd rather leave that to power manager applets that are exactly dedicated to monitoring battery status.

souravdas142, 2022-09-21:
> Points 1 and 2 are a bug in main, fixed in https://github.com/blueman-project/blueman/pull/1862.

Thank you, that MR solved the problem.

> Regarding point three, there's a suggestion to add the battery percentage to the tray icon tooltip in https://github.com/blueman-project/blueman/issues/1797. While that makes a lot of sense to me, the disconnect menu items seem like the wrong place to me.

Wow it really makes sense than putting battery icons beside disconnect items. But need to put important information only with device name, I have seen #1797 which suggest more impractical information.

> Point 4 is possible, yes, but I'd rather leave that to power manager applets that are exactly dedicated to monitoring battery status.

Nice to hear  that it is possible, thats mean battery will not go down fully (At least Blueman will continue to inform with battery discharge information) while watching movie. and saves more battery cycle (Yeah it is dependent on user, user should disconnect before battery dies) . Yeah it should be implemented then.

Thank you.

PR #1862 (2022-09-21): Fix per-device applet menu items
With #1847 priorities are expected to be unique. Otherwise the items get merged into one, so that there is only one disconnect and only one audio profile item as no explicit sub-indexes got added there.

Fixes #1861
commit 6df1bf74 (2022-09-21): Fix per-device applet menu items

With #1847 priorities are expected to be unique. Otherwise the items get merged into one, so that there is only one disconnect and only one audio profile item as no explicit sub-indexes got added there.

Fixes #1861
commit fff64d3f (2022-10-06): Fix per-device applet menu items

With #1847 priorities are expected to be unique. Otherwise the items get merged into one, so that there is only one disconnect and only one audio profile item as no explicit sub-indexes got added there.

Fixes #1861
closed 2022-09-21 by cschramm