blueman-project/blueman #1864: blueman-manager fails to start from tray icon

fdservices, 2022-09-27:
bluez 5.65-3
blueman 2.3.2-1
arch linux 5.19.10-arch1-1

Bluetooth experimental is set to false

Starting blueman-manager from a terminal produces some error output, **before its starts correctly**

<code>Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 518, in on_finished
    callback()<p>
  File "/usr/lib/python3.10/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 308, in <lambda>
    self._prepare_fader(row_fader, lambda: self.__fader_finished(device))<p>
  File "/usr/lib/python3.10/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 312, in __fader_finished
    super().device_remove_event(device)<p>
  File "/usr/lib/python3.10/site-packages/blueman/gui/DeviceList.py", line 175, in device_remove_event
    assert tree_iter is not None<p>
AssertionError</code>

I suspect that it is this error that causes the problem.

infirit, 2022-09-27:
I can't replicate this exact issue but I can reproduce it in another way. Quick fix (not solution) is below

```python

diff --git a/blueman/gui/DeviceList.py b/blueman/gui/DeviceList.py
index a0af7e06..37f757fd 100644
--- a/blueman/gui/DeviceList.py
+++ b/blueman/gui/DeviceList.py
@@ -172,7 +172,9 @@ class DeviceList(GenericList):
     def device_remove_event(self, device: Device) -> None:
         logging.debug(device)
         tree_iter = self.find_device(device)
-        assert tree_iter is not None
+
+        if tree_iter is None:
+            return
 
         if self.compare(self.selected(), tree_iter):
             self.emit("device-selected", None, None)
```

cschramm, 2022-09-28:
I don't think that error is a problem as it's in the context of the `animation-finished` event and the event handler can cope with such errors so that they are non-fatal. Thus blueman-manager starts correctly anyway.

Starting from the tray icon is generally possible in two way: Left-click on the icon (depends on the icon implementation and server) or via the "devices" item in its (right-click) menu. I assume that both ways do not work for you, correct?

To get behind that problem, you can

- check `~/.xsession-errors` for errors.
- try the left-click mechanism manually e.g. with `dbus-send --print-reply --dest=org.blueman.Applet /org/blueman/Applet org.blueman.Applet.Activate`.
- run `blueman-applet --loglevel debug` and / or `blueman-tray` from a terminal to see if they show any insights when trying to invoke `blueman-manager`.

cschramm, 2022-09-28:
As you didn't mention your desktop environment... you should probably check #1636.

infirit, 2022-09-28:
I can replicate this error by switching adapters. I traced it to `DeviceList.find_device`. It looks up the row in the `path_to_row` which in my case has already been populated with the rows from the other adapter and then just returns `None`. I haven't gotten to the bottom of it.

But yeah, it's not fatal so please look in the places @cschramm mentioned for some hints.

fdservices, 2022-09-28:
@cschramm  .xsession-errors is empty
did not know about left-click, same result
`$ dbus-send --print-reply --dest=org.blueman.Applet /org/blueman/Applet org.blueman.Applet.Activate
method return time=1664355316.313155 sender=:1.22 -> destination=:1.49 serial=131 reply_serial=2`
blueman-tray does nothing
blueman-applet run from the command line gives on right-click > devices and on left-click
`$ blueman-applet
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/blueman/main/DBusProxies.py", line 52, in call_finish
    proxy.call_finish(resp)
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4)
`
My normal DE is Openbox. Also does not work using Compiz standalone. Works correctly with XFCE4

I had already read those threads and checked my environment settings :)

@infirit I do have two adapters, disconnecting one of them does not seem to have any effect

fdservices, 2022-09-28:
@infirit ... but after a reboot, with only one adapter, it seems to have started working correctly?

cschramm, 2022-09-28:
Sounds exactly like #1541 to me.

> If you face the same issue, please check that `echo $DISPLAY` points to the correct value resp. any value at all.

> Looks like everything works fine if I run `dbus-update-activation-environment DISPLAY` upfront.

fdservices, 2022-09-28:
@infirit ... after a reboot, with two adapters, it seems to have stopped working again

@cschramm I had already read those threads and checked my environment settings - including the DISPLAY variable.

cschramm, 2022-09-28:
Alright, so, it seems like blueman-manager does not start correctly when invoked from the D-Bus server. I think you should see its (error level) output in `journalctl --user -u blueman-manager.service`.

fdservices, 2022-09-28:
@cschramm `-- No entries --`

fdservices, 2022-09-28:
This might help?
<code># systemctl status blueman-mechanism
○ blueman-mechanism.service - Bluetooth management mechanism
     Loaded: loaded (/usr/lib/systemd/system/blueman-mechanism.service; disabled; preset: disabled)
     Active: inactive (dead)

Sep 28 11:27:37 acer systemd[1]: blueman-mechanism.service: Deactivated successfully.
Sep 28 11:41:16 acer systemd[1]: Starting Bluetooth management mechanism...
Sep 28 11:41:17 acer systemd[1]: Started Bluetooth management mechanism.
Sep 28 11:42:08 acer systemd[1]: blueman-mechanism.service: Deactivated successfully.
Sep 28 11:51:19 acer systemd[1]: Starting Bluetooth management mechanism...
Sep 28 11:51:20 acer systemd[1]: Started Bluetooth management mechanism.
Sep 28 11:51:51 acer systemd[1]: blueman-mechanism.service: Deactivated successfully.
Sep 28 12:08:05 acer systemd[1]: Starting Bluetooth management mechanism...
Sep 28 12:08:06 acer systemd[1]: Started Bluetooth management mechanism.
Sep 28 12:08:37 acer systemd[1]: blueman-mechanism.service: Deactivated successfully.
</code>

<code>$ systemctl --user  status blueman-applet
× blueman-applet.service - Bluetooth management applet
     Loaded: loaded (/usr/lib/systemd/user/blueman-applet.service; static)
     Active: failed (Result: exit-code) since Wed 2022-09-28 12:06:11 CEST; 10min ago
    Process: 22069 ExecStart=/usr/bin/blueman-applet (code=exited, status=1/FAILURE)
   Main PID: 22069 (code=exited, status=1/FAILURE)
        CPU: 533ms

Sep 28 12:06:11 acer blueman-applet[22069]:   File "/usr/bin/blueman-applet", line 40, in <module>
Sep 28 12:06:11 acer blueman-applet[22069]:     app = BluemanApplet()
Sep 28 12:06:11 acer blueman-applet[22069]:   File "/usr/lib/python3.10/site-packages/blueman/main/Applet.py", line 19, in __init__
Sep 28 12:06:11 acer blueman-applet[22069]:     setup_icon_path()
Sep 28 12:06:11 acer blueman-applet[22069]:   File "/usr/lib/python3.10/site-packages/blueman/Functions.py", line 155, in setup_ico>
Sep 28 12:06:11 acer blueman-applet[22069]:     ic.prepend_search_path(ICON_PATH)
Sep 28 12:06:11 acer blueman-applet[22069]: AttributeError: 'NoneType' object has no attribute 'prepend_search_path'
Sep 28 12:06:11 acer systemd[582]: blueman-applet.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 12:06:11 acer systemd[582]: blueman-applet.service: Failed with result 'exit-code'.
Sep 28 12:06:11 acer systemd[582]: Failed to start Bluetooth management applet.
</code>

<code>$ systemctl --user  status blueman-manager
× blueman-manager.service - Bluetooth Manager
     Loaded: loaded (/usr/lib/systemd/user/blueman-manager.service; static)
     Active: failed (Result: exit-code) since Wed 2022-09-28 12:17:18 CEST; 23s ago
   Duration: 7ms
    Process: 29422 ExecStart=/usr/bin/blueman-manager (code=exited, status=1/FAILURE)
   Main PID: 29422 (code=exited, status=1/FAILURE)
        CPU: 573ms

Sep 28 12:17:17 acer systemd[582]: Starting Bluetooth Manager...
Sep 28 12:17:18 acer systemd[582]: Started Bluetooth Manager.
Sep 28 12:17:18 acer blueman-manager[29422]: cannot open display: 
Sep 28 12:17:18 acer systemd[582]: blueman-manager.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 12:17:18 acer systemd[582]: blueman-manager.service: Failed with result 'exit-code'.
</code>

<code>
$ echo $DISPLAY
:0
</code>

cschramm, 2022-09-28:
You do see messages in `systemctl --user status blueman-manager` but not `journalctl --user -u blueman-manager`? That's weird.

Anyway, `cannot open display:` is exactly what's to be expected from an invalid `$DISPLAY`. Did you try `dbus-update-activation-environment DISPLAY`? It can cause the problem on my side easily:

```
cschramm@y ~> DISPLAY= dbus-update-activation-environment DISPLAY
cschramm@y ~> dbus-send --dest=org.blueman.Applet /org/blueman/Applet org.blueman.Applet.Activate
cschramm@y ~> journalctl --user -u blueman-manager.service | tail -5
Sep 28 12:06:12 y systemd[752]: Starting Bluetooth Manager...
Sep 28 12:06:12 y systemd[752]: Started Bluetooth Manager.
Sep 28 12:06:12 y blueman-manager[71728]: cannot open display:
Sep 28 12:06:12 y systemd[752]: blueman-manager.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 12:06:12 y systemd[752]: blueman-manager.service: Failed with result 'exit-code'.
```

infirit, 2022-09-28:
It actually does require it but on a terminal the environment is already setup properly.

@cschramm is there something we can do in the unit file? I couldn't find anything useful other than check is an ENV variable is set.

fdservices, 2022-09-28:
@infirit `It actually does require it but on a terminal the environment is already setup properly.`

It also starts fine from a menu (.desktop) entry

cschramm, 2022-09-28:
I don't think there's much we can do about it. It seems like it's a rather rare case that D-Bus invokes a graphical application. :shrug: systemd does provide the `graphical-session.target` for this but it only indicates that a session is running, not that the environment is set up properly. We expect a proper setup and there is a proper setup in 99.99 % of all cases.
 We could of course catch the problem (i.e. blueman-applet could detect that blueman-manager does not work due to an inoperable display setup) and tell the user but, well... I pointed out the problem in the addendum to my very first comment here and we still made quite some rounds until @fdservices accepted the solution which I even quoted explicitly in my very next comment, so I'm not sure if that would make things much better. :see_no_evil: 

I don't really get why things go wrong in the first place, see https://github.com/blueman-project/blueman/issues/1636#issuecomment-980077862. My conclusion for X was more or less that people must be putting insufficient `xinitrc` files in place that do not include `/etc/X11/xinit/xinitrc.d` and thus miss their applications' intended configuration, especially [/etc/X11/xinit/xinitrc.d/50-systemd-user.sh](https://github.com/systemd/systemd/blob/main/xorg/50-systemd-user.sh), but do not compensate that with any custom configuration either. That's my proposed explanation for all cases with i3, bspwm and now Openbox, but as affected users always just reported that they had some misconfiguration or that an existing solution fixed their problem, I do not know what causes they actually had.

For Sway on Arch Linux, the other bunch of cases that I've seen, the explanation could actually be similar: A custom `~/.config/sway/config` that lacks the standard `include /etc/sway/config.d/*` and thus Arch Linux' `/etc/sway/config.d/50-systemd-user.conf` that would set up everything right.

fdservices, 2022-09-28:
I really don't think this is going anywhere. We have established that blueman-manager apparently runs correctly from anywhere - except the blueman-applet - and on certain DE's. We seem to know why it does not run from the applet, but we have no fix. OK.

Thanks for all the replies.

cschramm, 2022-09-28:
I actually do not know why the environment is not set up properly. The standard setup is: You have an xinitrc in place that includes `/etc/X11/xinit/xinitrc.d/*` and thus `/etc/X11/xinit/xinitrc.d/50-systemd-user.sh` and that script handles setting `DISPLAY` so that it matches the session.

infirit, 2022-09-28:
Can you try https://github.com/infirit/blueman/commit/23bcfcf3a13600610be7ed40db5ea064993f8cc6 and let us know if that helps. If you made any changes to fix this before please undo them.

infirit, 2022-09-28:
PR with all of the apps updated #1865

fdservices, 2022-09-29:
I feel a bit like I am drowning here. However, I compiled infirit@23bcfcf and the only real difference I could see was that left-click on the blueman-applet brought up the applet menu. Right-click also brought up the menu as exected. Click on dDevices showed the same error. Should I do something with the PR?

fdservices, 2022-10-01:
I am really sorry to have wasted your time.
@cschramm was absolutely right all along
`I actually do not know why the environment is not set up properly. The standard setup is: You have an xinitrc in place that includes /etc/X11/xinit/xinitrc.d/* and thus /etc/X11/xinit/xinitrc.d/50-systemd-user.sh and that script handles setting DISPLAY so that it matches the session.`
I was starting Openbox incorrectly so that .xinitrc was never executed. Setting DISPLAY and XAUTHORITY therefore had no effect since 50-systemd-user.sh was never run. I had read all the comments but it did not click that startx does not always run xinitrc. @cschramm  kept trying to tell me where the problem was, but I was just too stupid (again).

closed 2022-10-01 by fdservices

cschramm, 2022-10-01:
No worries, you're welcome. So what was the actual problem? Did you call `startx` with client options so that it considered xserverrc instead of xinitrc? 🤔 That seems to be another source for errors which I just learned about from the man page. 😄

fdservices, 2022-10-01:
Yes, so xinitrc never ran. Would it be an idea to test the dbus variables or simply run dbus-update-activation-environment  on start up? It does not seem to cause a problem if it is run more than once. Of course it should not be necessary, but it does seem to have caused a number of problems in the past.

cschramm, 2022-10-01:
You mean us, i.e. the blueman-applet process? :thinking: Not sure if that's a good idea as changing that environment is a rather unexpected side effect, but sure, we can even do it directly via D-Bus like:

```python
from gi.repository import Gio, GLib
import os

Gio.bus_get_sync(Gio.BusType.SESSION).call_sync(
    "org.freedesktop.DBus",
    "/org/freedesktop/DBus",
    "org.freedesktop.DBus",
    "UpdateActivationEnvironment",
    GLib.Variant("(a{ss})", ({"DISPLAY": os.environ["DISPLAY"]},)),
    None,
    Gio.DBusCallFlags.NONE,
    -1
)
```

(There does not seem to be an interface to read the environment so that we could pre-detect the problem and either decide to fix the environment or inform the user.)

cschramm, 2022-10-09:
@infirit: What do you think? Update the activation environment ourselves, show some message when it seems to be broken, or leave everything as it is?

infirit, 2022-10-09:
A big(ish) warning message is I think appropriate here. Maybe hint at what usually handles this so people can google for it?

cschramm, 2022-10-10:
See #1880. Input to make it more useful appreciated. :)

fdservices, 2022-10-10:
I know this is now off-topic, but I really do not understand why, if I run blueman-manager from the command line it works fine, if I run blueman-adapters from the command line it works fine. If I select Adaptors from the tray menu, blueman-adapter runs fine but if I select Devices from the  tray menu blueman-manager requires the dbus environment in order to run.

cschramm, 2022-10-10:
That is a reasonable question.

High level answer: If you run anything from the command line, it takes your command line's environment (especially its `DISPLAY`) which seems to be good then. If you run `blueman-manager` from the tray menu it does get activated via D-Bus, so that it takes your D-Bus activation environment. If you run anything else from the tray menu (besides `blueman-adapters` there's also `blueman-sendto` and `blueman-services`), it gets invoked from the `blueman-applet` process directly and thus takes its enviroment which must be good as blueman-applet itself would not run properly otherwise.

I actually cannot give the lower level answer on why do we use D-Bus activation for blueman-manager. It comes from [#1304](https://github.com/blueman-project/blueman/pull/1304/commits/3afc3b91294b61d5e4d5669c1ae799bdc5f75235), so @infirit can definitely say more.

infirit, 2022-10-11:
I'm not sure what people expect to happen when DISPLAY is unset under Xorg but any Gtk application will die even blueman-adapters. And this is the core of the problem, the DISPLAY environment variable is not available to the dbus user session due to a misconfigured system. Sure we can revert to the old way of using launching like we do with blueman-adapters but at some point you'll just run into another application that expect this to be setup properly.

```
[sander@infirit blueman]$ unset DISPLAY
[sander@infirit blueman]$ LANG=C blueman-adapters

(blueman-adapters:2700): Gtk-CRITICAL **: 17:59:23.584: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(blueman-adapters:2700): Gtk-CRITICAL **: 17:59:23.584: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(blueman-adapters:2700): Gtk-CRITICAL **: 17:59:23.584: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentatiefout (geheugendump gemaakt)
```

fdservices, 2022-10-11:
`at some point you'll just run into another application that expect this to be setup properly.`

Sure, but the only place that this caused me a problem was with blueman. Doesn't the KISS principle apply here?

fdservices, 2022-10-11:
PS: you unset the system-wide environment variable - not the dbus environment variable

infirit, 2022-10-11:
> PS: you unset the system-wide environment variable - not the dbus environment variable

Same thing. DISPLAY is not available to dbus when it launches blueman-manager, nothing good can come from it. I was just illustrating that no X application will run if it doesn't have know which DISPLAY to use.

> Sure, but the only place that this caused me a problem was with blueman. Doesn't the KISS principle apply here?

No, we should be able to rely that the distribution (or user if they decide to configure thing themselves) has setup the proper environment. This isn't something new, distributions have been doing this properly for over a decade.

The only reason I would consider changing it is to just get rid of the complaints.

fdservices, 2022-10-12:
Clearly if dbus is required for blueman to function properly, then it is reasonable to leave it as is and to expect dbus to be correctly installed.

On the other hand if it is only used to launch blueman-manager from the tray icon, then `I would consider changing it is to just get rid of the complaints`