blueman-project/blueman #1868: mt7921au chipset

LargePrime, 2022-09-30:
META NOTE;
Perhaps a command or note on how to retrieve these version numbers would be useful to complete this template.

blueman:2.3.2 
 blueman-applet 13.39.57
BlueZ:
Distribution: Linux 5.19.12-artix1-1 artix-lxqt-openrc artixlinux.org 
Desktop environment:LXQt OpenRC

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

lsusb
Bus 009 Device 005: ID 0e8d:7961 MediaTek Inc. Wireless_Device
bluetoothctl list
Controller E0:E1:A9:34:A4:66 Compast [default]

python3
Python 3.10.7 (main, Sep  7 2022, 18:52:02) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from blueman.gui.Notification import Notification
>>> Notification("Title", "Message", 0, [["action1", "Action 1"], ["action2", "Action 2"]]).show()
Message shows

bluetoothctl below
https://gist.github.com/LargePrime/1e6591fed7c53358a46793510f25240d
I did not do a pair command because I do not know the syntax or device address

Device will not pair.

Extra info
https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md
suggests this device is supported.

WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin
BT_RAM_CODE_MT7961_1_2_hdr.bin
are in /lib/firmware/mediatek and were put there by the distro.

Thank you for your time, and sorry for whatever I missed.

cschramm, 2022-09-30:
> Device will not pair.

Is this the problem statement? I assume you're scanning, discovering some specific device and then use blueman to pair it but ... what? An error message?

LargePrime, 2022-09-30:
Scan finds no devices.
Devices are available and are seen with other bluetooth devices

cschramm, 2022-10-09:
So the adapter is registered and seems to be scanning properly but it does not find any devices. Not sure if I can help here but I'd enable BlueZ debug logging and check what the process gives there and also check kernel messages of course. If everything looks good there, I do not have another idea.

LargePrime, 2022-11-26:
can you point me at a thing or walk me through that?

cschramm, 2022-11-26:
https://wiki.archlinux.org/title/bluetooth#Debugging

And `dmesg -w` for following kernel messages.

When both are running in terminals, try scanning and check what's showing up.

closed 2023-02-03 by github-actions[bot]

LargePrime, 2023-02-04:
bluetoothctl list
Controller E0:E1:A9:34:A4:66 Compast [default]
Controller 00:1A:7D:DA:71:13 box
lsusb
Bus 004 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 009 Device 004: ID 0e8d:7961 MediaTek Inc. Wireless_Device

I physically removed the dongle called box

killall blueman-applet
blueman-applet --loglevel debug
```
blueman-applet 11.25.12 INFO     PluginManager:87 load_plugin: ['TransferService', 'StatusNotifierItem', 'AuthAgent', 'ExitItem', 'SerialManager', 'Networking', 'PowerManager', 'StandardItems', 'GameControllerWakelock', 'Menu', 'PPPSupport', 'AutoConnect', 'NMPANSupport', 'ConnectionNotifier', 'StatusIcon', 'NetUsage', 'DBusService', 'KillSwitch', 'NMDUNSupport', 'ShowConnected', 'RecentConns', 'DhcpClient', 'DiscvManager', 'DisconnectItems']
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 11.25.12 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci1 {'Address': 'E0:E1:A9:34:A4:66', 'AddressType': 'public', 'Name': 'box', 'Alias': 'Compast', 'Class': 8126724, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0542', 'Roles': ['central', 'peripheral']}
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 11.25.12 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 11.25.12 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 11.25.12 INFO     KillSwitch:106 io_event  : killswitch registered 2
blueman-applet 11.25.12 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 11.25.12 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 11.25.12 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 11.25.12 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.31
blueman-applet 11.25.12 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DisconnectItems.DisconnectItems object at 0x7f927d9070d0>
[... cut ...]
I physically turn on known good device and set to pairing. restart pairing in applet
```

blueman-applet 11.27.45 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci1 {'Discovering': True}
blueman-applet 11.27.45 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering True
blueman-applet 11.27.55 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci1 {'Discovering': False}
blueman-applet 11.27.55 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering False
```

python
Python 3.10.9 (main, Dec 25 2022, 21:29:15) [GCC 12.2.0] on linux

cschramm, 2023-02-08:
Nothing that sticks out to me, but that does not include a bluetoothd debug log of the scan. Regarding the kernel messages I'd expect at least some messages from `mt7921u`, although that driver does not seem to be involved in Bluetooth.

LargePrime, 2023-02-10:
how do i get a " bluetoothd debug log of the scan"

cschramm, 2023-02-25:
BlueZ successfully runs a complete discovery and finds absolutely nothing, so if there is a responsive device in reach, the chip does not seem to work as expected. I don't know how to get behind this. It's probably best to address the BlueZ project or Linux.