blueman-project/blueman #1870: Duplicate tray icon

clefebvre, 2022-10-06:
blueman: 2.3.2
BlueZ: 5.64-0ubuntu1
Distribution: Mint 21
Desktop environment: Cinnamon 5.4

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

This is an issue which seems to happen randomly. We've seen people talk about it on the Mint forums. I got lucky here on this computer, I have it in a way that it seems to happen all the time on every reboot.

Blueman provides two tray icons in the Cinnamon panel:

[image]

The StatusItemNotifier plugin is enabled:

[image]

Both the applet and the tray processes are running:

[image]

Here are the settings used by blueman:

[image]

@mtwebster the left tray icon has the following characteristics:

- Its tooltip is "Blueman"
- Its context menu is blank
- It shows/hides the Blueman manager on left click
- It's provided by XappStatusApplet (I can see that by pressing Ctrl+Right-Click -> About...)

The right tray icon seems to be an xembed:

- Its tooltip is "Bluetooth Enabled"
- Its context menu is properly populated
- It shows/hide the Blueman manager on left click
- Cinnamon doesn't show which applet provides it.... when clicking Ctrl+Right-Click, we still get the Blueman context menu.

Cinnamon's logs:

[image]

We see both an XappStatusIcon and a systray here:

```
info t=2022-10-06T12:30:08Z Adding XAppStatusIcon: blueman (:1.47/org/x/StatusIcon/Icon_1)
info t=2022-10-06T12:30:08Z Adding systray: blueman-tray (24x24px)
```
I get the feeling this is an issue in Cinnamon's XappStatus applet, but it's possible there's something dodgy in the way Blueman switches between AppIndicator and GtkStatusIcon.

clefebvre, 2022-10-06:
I went ahead and gave the Blueman XDG autostart a 20s delay and the problem no longer happens. It definitely looks like a race condition.

clefebvre, 2022-10-06:
Correction.. I don't think _on_name_appeared() is called again. It looks like the StatusNotifierPlugin simply registers again whenever the SNI bus name appears:

[image]

I think there are multiple issues at play here:

- In Cinnamon, blueman.desktop seems to be launched before SNI is ready. xapp-sn-watcher should be finished setting things up. This race condition shouldn't happen.
- In Blueman, we watch for SNI to pop up on DBUS but by the time it does we already started a GtkStatusIcon and we end up with two concurrent implementations. We should either not fallback (and only do SNI when the plugin is active), or have a timeout in place before which we don't fallback and after which we kill SNI support.

cschramm, 2022-10-06:
Yes, you're absolutely right. #1801 was half-baked. It only covers the case correctly where the item initially got registered but the watcher gets restarted or replaced afterward. It's incorrect if the watcher becomes available too late.

closed 2022-10-06 by cschramm

pacho2, 2024-12-02:
Hello, 

We are still hitting this issue with blueman-2.4.3 on Gnome with ubuntu/gnome-shell-extension-appindicator extension and, now (version 59 of the extension), it is the same if we revert the commit referred at https://github.com/ubuntu/gnome-shell-extension-appindicator/issues/419 or not.

Adding some delay to the start looks to work, but I don't understand why it happens with blueman and not other apps :/

Thanks for your help

cschramm, 2024-12-02:
@pacho2: That is very probably not this issue. This issue is about icons being shown for both a StatusNotifierItem and an XEmbed fallback at the time in blueman 2.3. The issue you mean is about two icons being shown for a single StatusNotifierItem, was reported by a user of that Gnome Shell extension in #419, turned out to be caused by the extension itself, and reverting a specific commit fixed it. This was last confirmed in January, so presumably for version 57 of the extension, as it seems like the reversion of the problematic commit never got upstreamed. If the patch started failing for version 58, the inner workings of the extension probably changed in some way. If someone could tell us why the extension acts as it acts, we could look into a mitigation from the blueman side.

pacho2, 2024-12-02:
The current issue I am seeing is that, at login time, two XEmbed icons are shown and Indicators are not used. If I restart the shell, only one (using Indicators) is shown. Maybe this is caused by blueman starting before the shell extension for indicator support is loaded.

cschramm, 2024-12-04:
So this is clearly not this issue. It is not the gnome-shell-extension-appindicator issue either then, as that extension shows StatusNotifierItems, not XEmbed icons. Please open a new issue to discuss what you're seeing.