blueman-project/blueman #1873: Blueman Manager 2.3.2

felixecited, 2022-10-07:
blueman: 2.3.2
BlueZ:
Distribution: Linux Mint 21
Desktop environment: Cinamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2022-10-07:
It looks like the kernel driver for the RTL8821CE bluetooth adapter is crashing. Which is likely the reason it doesn't want to reconnect.. Does it still happen if you use `bluetoothctl connect <bluetoothaddress>`?

yuquanguo, 2022-10-08:
I have the same problem, I have K480 3 device keyboard. It outputs an error "Failed to connect: org.bluez.Error.Failed br-connection-create-socket" if i use "bluetoothctl connect <bluetoothaddress>"

cschramm, 2022-10-09:
@yuquanguo: Are you saying that you're on rtl8821ce and see it crashing as well or that you see the same behavior without such a driver crash? You can check the bluetoothd logs, enabling debug logging, to get behind where the error is happening.

felixecited, 2022-10-13:
I just received an Blueman update to 2.3.3 k380 still disconnects when switching to phone or tablet then back. M720 mouse has no problem its always connected across all devices

felixecited, 2022-12-14:
what driver is crashing? i am not providing drivers. all drivers are included in linux mint. if some driver is crashing it needs updating and my system if fully updated

infirit, 2022-12-14:
> what driver is crashing? i am not providing drivers. all drivers are included in linux mint. if some driver is crashing it needs updating and my system if fully updated

Never said you provided the driver, just that it was crashing. Bugs can be present even if you have a fully updated install.

Regardless, crashing drivers is not something we can fix. Blueman is just the frontend to the underlying Bluetooth stack. Perhaps posting the kernel log on the mint forums helps with solving the problem.

closed 2023-02-20 by github-actions[bot]