blueman-project/blueman #1874: Fedora, i3, blueman-applet fails to start

artemave, 2022-10-07:
blueman: 2.3.2
BlueZ: 5.6.5
Distribution: Fedora 36 (Workstation Edition)
Desktop environment: i3wm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I am setting up a new laptop and having trouble starting blueman-applet:

```
❯ blueman-applet
blueman-applet 03.11.07 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 03.11.07 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 03.11.32 ERROR    BasePlugin:83 _load     : Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/blueman/main/DBusProxies.py", line 27, in __init__
    self.init()
gi.repository.GLib.GError: g-io-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached (24)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/blueman/plugins/BasePlugin.py", line 77, in _load
    self.on_load()
  File "/usr/local/lib/python3.10/site-packages/blueman/plugins/applet/Networking.py", line 28, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.10/site-packages/blueman/plugins/applet/Networking.py", line 49, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.10/site-packages/blueman/gobject.py", line 11, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/blueman/main/DBusProxies.py", line 34, in __init__
    super().__init__(name='org.blueman.Mechanism', interface_name='org.blueman.Mechanism',
  File "/usr/local/lib/python3.10/site-packages/blueman/main/DBusProxies.py", line 29, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached
```

I am a pretty inexperienced user, please let me know if I forgot to add some obvious detail or indeed missed something obvious. Thanks a lot!

cschramm, 2022-10-07:
Please check `journalctl -u blueman-mechanism.service` for errors.

artemave, 2022-10-07:
```
❯ journalctl -u blueman-mechanism.service
Oct 06 11:08:01 fedora systemd[1]: Starting blueman-mechanism.service - Bluetooth management mechanism...
Oct 06 11:08:31 fedora systemd[1]: blueman-mechanism.service: Deactivated successfully.
Oct 06 11:08:31 fedora systemd[1]: Started blueman-mechanism.service - Bluetooth management mechanism.
-- Boot 427000cbec2c4aaa887daea6b128f363 --
Oct 06 11:39:50 fedora systemd[1]: Starting blueman-mechanism.service - Bluetooth management mechanism...
Oct 06 11:40:20 fedora systemd[1]: blueman-mechanism.service: Deactivated successfully.
Oct 06 11:40:20 fedora systemd[1]: Started blueman-mechanism.service - Bluetooth management mechanism.
```

artemave, 2022-10-07:
It's suddenly working. I restarted the laptop, but I am pretty sure I've done that before. Sorry for disturbing and thanks for your help!

closed 2022-10-07 by artemave