blueman-project/blueman #1876: Connecting to a device doesn't work at first clicks, only after right click

sudoaza, 2022-10-07:
blueman: 2.3.1
BlueZ: 5.65-2
Distribution: Manjaro
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When I try to connect to headphones I double click on the device in the blueman-manager and nothing happens, even after a couple of tries.
Then I right click and get a small popup with the status "Connecting..." (?!) 
Up untill here none of the interactions produced any log on blueman.
But after the right click a double left click will start the connection with the progress bar showing bellow and connecting successfully. 

Expected outcome:

Double click connects to device without having to right click before.

infirit, 2022-10-07:
Relevant trace.
```python
blueman-manager 12.25.53 INFO     ManagerDeviceMenu:103 set_op    : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f663c79fd40 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x55c89b05d100)>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 239, in on_event_clicked
    self.menu.connect_service(row["device"])
  File "/usr/lib/python3.10/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 141, in connect_service
    self.set_op(device, _("Connecting…"))
  File "/usr/lib/python3.10/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 104, in set_op
    if inst.SelectedDevice == self.SelectedDevice and not (inst.is_popup and not inst.props.visible):
AttributeError: 'ManagerDeviceMenu' object has no attribute 'SelectedDevice'
```

infirit, 2022-10-07:
So not all instances of the menu have been generated in this situation. We can either check if we have a `SelectedDevice` class atribute or generate all instances. Below is the former approach.

```patch
diff --git a/blueman/gui/manager/ManagerDeviceMenu.py b/blueman/gui/manager/ManagerDeviceMenu.py
index 47077899..eea9de0f 100644
--- a/blueman/gui/manager/ManagerDeviceMenu.py
+++ b/blueman/gui/manager/ManagerDeviceMenu.py
@@ -100,6 +100,8 @@ class ManagerDeviceMenu(Gtk.Menu):
     def set_op(self, device: Device, message: str) -> None:
         ManagerDeviceMenu.__ops__[device.get_object_path()] = message
         for inst in ManagerDeviceMenu.__instances__:
+            if not hasattr(inst, "SelectedDevice"):
+                return
             logging.info(f"op: regenerating instance {inst}")
             if inst.SelectedDevice == self.SelectedDevice and not (inst.is_popup and not inst.props.visible):
                 inst.generate()
@@ -113,6 +115,8 @@ class ManagerDeviceMenu(Gtk.Menu):
     def unset_op(self, device: Device) -> None:
         del ManagerDeviceMenu.__ops__[device.get_object_path()]
         for inst in ManagerDeviceMenu.__instances__:
+            if not hasattr(inst, "SelectedDevice"):
+                return
             logging.info(f"op: regenerating instance {inst}")
             if inst.SelectedDevice == self.SelectedDevice and not (inst.is_popup and not inst.props.visible):
                 inst.generate()
```