blueman-project/blueman #1887: Cannot connect any sound device

schottz, 2022-10-13:
After last update, I'm unable to connect to my sound devices. Both my Alexa and JBL Headphone, even listed on discovered devices, cannot be paired.

The error like below is shown at the bottom of the window.

blueman: 2.3.3
BlueZ:
Distribution: Linux Mint 21
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

<script src="https://gist.github.com/schottz/e9fa4935cba889c0b71d022cf38effd3.js"></script>

cschramm, 2022-10-13:
Can you check #1888?

cschramm, 2022-10-14:
Would be great if you can test with the fix.

An easy way would be to download https://github.com/blueman-project/blueman/blob/1cb88bd8c05a0106d974eacba9bc984396d80c0d/blueman/plugins/applet/Menu.py and https://github.com/blueman-project/blueman/blob/1cb88bd8c05a0106d974eacba9bc984396d80c0d/blueman/main/indicators/StatusNotifierItem.py and replace the respective files on your system.

No worries if you're not able to test it.

schottz, 2022-10-14:
I’ll do and return

schottz, 2022-10-14:
Hi @cschramm . It worked! Replaced the files and restarted the computer. I had the error below at first. After turning the computer off and on again, it seems it's working well.

Thank you

`Connection Failed: org.freedesktop.DBus.Error.UnknownObject Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist`

cschramm, 2022-10-14:
Awesome, thanks for double-checking! A 2.3.4 is already published (here, not yet in Linux Mint).

PR #1888 (2022-10-14): Add missing multi-level ID support for menu items
1297d8806e416f90a4dd0bfcef6e76849ef65675 requires this to be available but it is not as it is part of #1652. This backports the relevant changes.

Fixes #1887.