blueman-project/blueman #1895: Reverse tethering (PAN) is broken in local services

foggy677, 2022-10-16:
blueman: 2.1.4
BlueZ: 5.55
Distribution: Debian 11
Desktop environment: GNOME 3.38.5

[ *] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The reverse tethering isn't working anymore with Debian 11 when I activate the NAP service in Blueman (with control trough NetworkManager as a default) "local services".
It used to work with older Debian version and with Ubuntu 16.04 LTS.
My smartphone Samung Galaxy A22 (Android 12) is given a bluetooth link and an IP adress from the PC, but there is no internet flow trough. A PAN (pan1) is created. I can ping the smartphone from the PC with $ping command.
Exactly the same problem occurs in Ubuntu 22.04 LTS.

Am I doing something wrong, or is there a bug?

Here an extract from /var/log/syslog:

1°) When activating in local services and connecting my smartphone:

Oct 16 10:41:09 damien-esprimop2560 systemd[1372]: Started Application launched by gnome-shell.
Oct 16 10:41:09 damien-esprimop2560 dbus-daemon[1394]: [session uid=1000 pid=1394] Activating via systemd: service name='org.blueman.Applet' unit='blueman-applet.service' requested by ':1.185' (uid=1000 pid=7614 comm="/usr/bin/python3 /usr/bin/blueman-manager ")
Oct 16 10:41:09 damien-esprimop2560 systemd[1372]: Starting Bluetooth management applet...
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]: blueman-applet 10.41.09 ERROR    PluginManager:91 load_plugin: Unable to load plugin module GameControllerWakelock
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]: Traceback (most recent call last):
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 89, in load_plugin
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:     importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:     return _bootstrap._gcd_import(name[level:], package, level)
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:   File "/usr/lib/python3/dist-packages/blueman/plugins/applet/GameControllerWakelock.py", line 14, in <module>
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]:     raise ImportError('This is not an X11 screen')
Oct 16 10:41:09 damien-esprimop2560 blueman-applet[7622]: ImportError: This is not an X11 screen
Oct 16 10:41:09 damien-esprimop2560 dbus-daemon[452]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.160' (uid=1000 pid=7622 comm="/usr/bin/python3 /usr/bin/blueman-applet ")
Oct 16 10:41:10 damien-esprimop2560 systemd[1]: Starting Bluetooth management mechanism...
Oct 16 10:41:10 damien-esprimop2560 blueman-mechanism[7632]: Unable to init server: Could not connect: Connection refused
Oct 16 10:41:10 damien-esprimop2560 blueman-mechanism[7632]: Unable to init server: Impossible de se connecter : Connection refused
Oct 16 10:41:10 damien-esprimop2560 blueman-mechani[7632]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Oct 16 10:41:10 damien-esprimop2560 dbus-daemon[452]: [system] Successfully activated service 'org.blueman.Mechanism'
Oct 16 10:41:10 damien-esprimop2560 systemd[1]: Started Bluetooth management mechanism.
Oct 16 10:41:10 damien-esprimop2560 dbus-daemon[1394]: [session uid=1000 pid=1394] Successfully activated service 'org.blueman.Applet'
Oct 16 10:41:10 damien-esprimop2560 systemd[1372]: Started Bluetooth management applet.
Oct 16 10:41:10 damien-esprimop2560 blueman-applet[7631]: blueman-tray version 2.1.4 starting
Oct 16 10:41:10 damien-esprimop2560 blueman-applet[7631]: There is an instance already running
Oct 16 10:41:10 damien-esprimop2560 blueman-tray[7628]: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed
Oct 16 10:41:18 damien-esprimop2560 systemd-udevd[7658]: Using default interface naming scheme 'v247'.
Oct 16 10:41:18 damien-esprimop2560 systemd-udevd[7658]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1293] manager: (pan1): new Bridge device (/org/freedesktop/NetworkManager/Devices/10)
Oct 16 10:41:18 damien-esprimop2560 avahi-daemon[448]: Joining mDNS multicast group on interface pan1.IPv4 with address 10.137.137.1.
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1691] device (pan1): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1745] device (pan1): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1758] device (pan1): Activation: starting connection 'pan1' (98c79a40-89ef-42d9-9df8-0c0f50c059a3)
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1766] device (pan1): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1789] device (pan1): state change: prepare -> config (reason 'none', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1828] device (pan1): state change: config -> ip-config (reason 'none', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.1839] device (pan1): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 avahi-daemon[448]: New relevant interface pan1.IPv4 for mDNS.
Oct 16 10:41:18 damien-esprimop2560 avahi-daemon[448]: Registering new address record for 10.137.137.1 on pan1.IPv4.
Oct 16 10:41:18 damien-esprimop2560 dbus-daemon[452]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.10' (uid=0 pid=453 comm="/usr/sbin/NetworkManager --no-daemon ")
Oct 16 10:41:18 damien-esprimop2560 systemd[1]: Starting Network Manager Script Dispatcher Service...
Oct 16 10:41:18 damien-esprimop2560 dbus-daemon[452]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Oct 16 10:41:18 damien-esprimop2560 systemd[1]: Started Network Manager Script Dispatcher Service.
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.2268] device (pan1): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.2271] device (pan1): state change: secondaries -> activated (reason 'none', sys-iface-state: 'external')
Oct 16 10:41:18 damien-esprimop2560 NetworkManager[453]: <info>  [1665909678.2284] device (pan1): Activation: successful, device activated.
Oct 16 10:41:18 damien-esprimop2560 dnsmasq[7684]: démarrage avec le DNS désactivé (version 2.85)
Oct 16 10:41:18 damien-esprimop2560 dnsmasq[7684]: options à la compilation : IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
[... 73 lines cut ...]
MASQUERADE  all  --  10.137.137.0/24      anywhere  

damien@damien-esprimop2560:~$ sudo iptables -t filter -L FORWARD
Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
[... cut ...]
damien@damien-esprimop2560:~$ cat /proc/sys/net/ipv4/ip_forward
1

damien@damien-esprimop2560:~$ ls /proc/sys/net/ipv4/conf/*/forwarding
/proc/sys/net/ipv4/conf/all/forwarding    /proc/sys/net/ipv4/conf/default/forwarding  /proc/sys/net/ipv4/conf/pan1/forwarding
/proc/sys/net/ipv4/conf/bnep0/forwarding  /proc/sys/net/ipv4/conf/enp3s0/forwarding
/proc/sys/net/ipv4/conf/bnep1/forwarding  /proc/sys/net/ipv4/conf/lo/forwarding

damien@damien-esprimop2560:~$ cat /proc/sys/net/ipv4/conf/*/forwarding
1
1
1
1
1
1
1

cschramm, 2022-10-16:
Sorry for my unresponsiveness in the Debian issue. Your response is still in my overflowing inbox... :sweat_smile: Thanks for forwarding it here.

What's additionally mentioned [there](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1021555) is a message on `br_netfilter` not being loaded as in #739.

The routing setup looks good from the data you provided.

Please check if `br_netfilter` is loaded (`lsmod`) and in case it's not try if `modprobe`ing it helps. I don't fully understand the warning and where it exactly comes from, but if I get it right, some `iptable` rules may get managed fine but do not actually have an effect without the module. That could explain why routing does not work even though the setup looks perfectly fine.

foggy677, 2022-10-16:
Hi,
lsmod returns that no br_netfilter module is loaded by default.
So i entered $ sudo modprobe br_netfilter and now lsmod returns:
damien@damien-esprimop2560:~$ lsmod | grep filter
br_netfilter           32768  0
bridge                258048  1 br_netfilter

So the module is now loaded. I restarted the whole process of desactivating/activating NAP  with "local services" in blueman. Then I linked my smartphone to my Debian desktop trough bluetooth. The bluetooth link is up , the internet with bluetooth icon appears on my smartphone, but there is still no internet flow :(
By the way, I still get the same message in /var/log/syslog:

Oct 16 19:28:45 damien-esprimop2560 kernel: [  510.533268] perf: interrupt took too long (3146 > 3145), lowering kernel.perf_event_max_sample_rate to 63500
Oct 16 19:28:49 damien-esprimop2560 kernel: [  514.650331] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Oct 16 19:28:49 damien-esprimop2560 kernel: [  514.652492] Bridge firewalling registered

With Ubuntu 16.04 LTS, my reverse tethering works fine and the same error message appears in that file, so I am very confused...

infirit, 2022-10-16:
Just putting it here as I had was having this on Gentoo some time ago #739. Let me setup a degian 11 vm and see if I can reproduce at least.

cschramm, 2022-10-16:
What I'd do first to get behind this is check in Wireshark if expected traffic from / to the client is visible on `pan1` / `bnep1` and the upstream connection.

(N.B.: I did not find the time to test the PAN service yet (I'm actually unsure if I do have a working client device), so I'm not saying that it does work in general.)

infirit, 2022-10-16:
No traffic on bnep0 outside of the odd ICMP6 router solicitation.

foggy677, 2022-10-17:
Thank you for your investigation.
Is this lack of trafic a bug or a configuration issue?

infirit, 2022-10-17:
I'll poke at it some more tonight. I haven't yet replicated on debian 11 but on my arch system. The iso has been downloaded so once I get the VM up I'll let you know. I'll also see if it can post some screenshots of how it works on my fairly stock Android phone, maybe it helps you setup.

infirit, 2022-10-17:
Ok, confirmed on debian 11. And yes it's related to not having dns as removing `--port=0` make things worl

infirit, 2022-10-17:
@cschramm This is how I make the phone use the internet over the bluetooth connection.
If I click on the device it shows me this screen where I can toggle internet on/off.
[image]

foggy677, 2022-10-20:
Hello,
I removed the --port=0 argument in /usr/lib/python3/dist-packages/blueman/main/NetConf.py, and now reverse tethering trough bluetooth works fine! Thank you for your help.
Can this fix be definitely introduced in the blueman debian (and ubuntu 22.04 LTS) package as a security release?

Here is a sample from my /var/log/syslog file, with reverse tethering working now:

Oct 20 10:24:22 damien-esprimop2560 dbus-daemon[448]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.129' (uid=1000 pid=3704 comm="/usr/bin/python3 /usr/bin/blueman-services ")
Oct 20 10:24:22 damien-esprimop2560 systemd[1]: Starting Bluetooth management mechanism...
Oct 20 10:24:23 damien-esprimop2560 blueman-mechanism[3711]: Unable to init server: Could not connect: Connection refused
Oct 20 10:24:23 damien-esprimop2560 blueman-mechanism[3711]: Unable to init server: Impossible de se connecter : Connection refused
Oct 20 10:24:23 damien-esprimop2560 blueman-mechani[3711]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Oct 20 10:24:23 damien-esprimop2560 dbus-daemon[448]: [system] Successfully activated service 'org.blueman.Mechanism'
Oct 20 10:24:23 damien-esprimop2560 systemd[1]: Started Bluetooth management mechanism.
Oct 20 10:24:23 damien-esprimop2560 kernel: [ 1369.298800] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Oct 20 10:24:23 damien-esprimop2560 systemd-udevd[3717]: Using default interface naming scheme 'v247'.
Oct 20 10:24:23 damien-esprimop2560 systemd-udevd[3717]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2112] manager: (pan1): new Bridge device (/org/freedesktop/NetworkManager/Devices/6)
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2368] device (pan1): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2423] device (pan1): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2446] device (pan1): Activation: starting connection 'pan1' (571a4582-5768-4a08-871b-0c7e95a0d130)
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2452] device (pan1): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2462] device (pan1): state change: prepare -> config (reason 'none', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 avahi-daemon[444]: Joining mDNS multicast group on interface pan1.IPv4 with address 10.126.205.1.
Oct 20 10:24:23 damien-esprimop2560 avahi-daemon[444]: New relevant interface pan1.IPv4 for mDNS.
Oct 20 10:24:23 damien-esprimop2560 avahi-daemon[444]: Registering new address record for 10.126.205.1 on pan1.IPv4.
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2503] device (pan1): state change: config -> ip-config (reason 'none', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2512] device (pan1): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 dbus-daemon[448]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=449 comm="/usr/sbin/NetworkManager --no-daemon ")
Oct 20 10:24:23 damien-esprimop2560 systemd[1]: Starting Network Manager Script Dispatcher Service...
Oct 20 10:24:23 damien-esprimop2560 dbus-daemon[448]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Oct 20 10:24:23 damien-esprimop2560 systemd[1]: Started Network Manager Script Dispatcher Service.
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2931] device (pan1): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2934] device (pan1): state change: secondaries -> activated (reason 'none', sys-iface-state: 'external')
Oct 20 10:24:23 damien-esprimop2560 NetworkManager[449]: <info>  [1666254263.2947] device (pan1): Activation: successful, device activated.
Oct 20 10:24:24 damien-esprimop2560 dnsmasq[3753]: demarré, version 2.85 (taille de cache 150)
Oct 20 10:24:24 damien-esprimop2560 dnsmasq[3753]: options à la compilation : IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Oct 20 10:24:24 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCP, plage d'adresses IP 10.126.205.2 -- 10.126.205.254, durée de bail 1h
Oct 20 10:24:24 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCP, sockets bound exclusively to interface pan1
Oct 20 10:24:24 damien-esprimop2560 dnsmasq[3753]: Lecture de /etc/resolv.conf
Oct 20 10:24:24 damien-esprimop2560 dnsmasq[3753]: utilise le serveur de nom 127.0.0.53#53
Oct 20 10:24:24 damien-esprimop2560 dnsmasq[3753]: lecture /etc/hosts - 5 adresses
[... cut ...]
Oct 20 10:25:10 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCPOFFER(pan1) 10.126.205.182 0c:02:bd:04:d7:5f 
Oct 20 10:25:10 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCPDISCOVER(pan1) 0c:02:bd:04:d7:5f 
Oct 20 10:25:10 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCPOFFER(pan1) 10.126.205.182 0c:02:bd:04:d7:5f 
Oct 20 10:25:10 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCPREQUEST(pan1) 10.126.205.182 0c:02:bd:04:d7:5f 
Oct 20 10:25:10 damien-esprimop2560 dnsmasq-dhcp[3753]: DHCPACK(pan1) 10.126.205.182 0c:02:bd:04:d7:5f A22-de-Mon-compte

infirit, 2022-10-20:
> Can this fix be definitely introduced in the blueman debian (and ubuntu 22.04 LTS) package as a security release?

I don't think they'll go for that. But I'll let chris comment on this point as chris maintains it in debian

cschramm, 2022-10-20:
No way. That's not how those distributions work. There are point releases to get minimal bug fixes into Debian stable but requirements are rather high and limited to severe bugs: https://www.debian.org/doc/manuals/developers-reference/pkgs.html#upload-stable I'd even challenge calling this behavioral change a bug at all.

cschramm, 2022-10-20:
> @cschramm This is how I make the phone use the internet over the bluetooth connection. If I click on the device it shows me this screen where I can toggle internet on/off.

My phone does not show the relevant toggle. No clue why, but I need a new one anyway. Probably a new point on the list of reasons. :see_no_evil:

foggy677, 2022-10-29:
Hello,
Since I have removed the --port=0 argument in /usr/lib/python3/dist-packages/blueman/main/NetConf.py, reverse tethering trough bluetooth works fine! I switched on permanently the NAP in the "local services" of Blueman.

But now an error window appears at the logging in Debian with my account after power up.
See the attached screen copy.
Blueman complains about the resolv.conf file that doesn't exist: this is normal, because my internet on the Debian box isn't up yet when I log in with my account, I set up Internet access manually after that logging in.

Can the python script be modified to take into account that fact?

[image]

closed 2023-01-06 by github-actions[bot]