blueman-project/blueman #1898: Blueman is not listing my bluetooth keyboard

viveksreedharan88, 2022-10-16:
Blueman is unable to list my bluetooth keyboard. Device is not showing up in the list. Please advise. 

blueman:
BlueZ:
Distribution: Linux Mint
Desktop environment:Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.
[image]

viveksreedharan88, 2022-10-20:
May I know what info is needed?

cschramm, 2022-10-20:
The blueman version would be a great start.

Do you see the device from another system?

Did you try disabling `Hide unnamed devices` in the `View` menu?

viveksreedharan88, 2022-10-30:
Blueman 2.3.4.  have unchecked the Hide unchecked devices in view menu. Afer the recent update, the keyboard is connecting and is working.Apologies for the delay in updating this!

closed 2022-10-30 by viveksreedharan88

infirit, 2022-11-29:
> Blueman 2.3.4. have unchecked the Hide unchecked devices in view menu. Afer the recent update, the keyboard is connecting and is working.Apologies for the delay in updating this!

@viveksreedharan88 Would you mind telling me which keyboard this was. It seems more people are having difficulty with hiding these by default.

viveksreedharan88, 2022-11-30:
No problem. Its Logitech K480 Wireless Multi-Device Keyboard.