blueman-project/blueman #1903: Launching blueman-manager over dbus is broken with default XFCE4 on Debian 11

infirit, 2022-10-18:
Me sad.

cschramm, 2022-10-18:
Are you trying https://blueman.cschramm.eu/debian/? Something seems wrong with the package. It just totally lacks org.blueman.Manager.service (but has org.blueman.Mechanism.conf twice). :see_no_evil:

infirit, 2022-10-18:
oh, then that may be it. I'll check tomorrow to be sure. Too late now.

cschramm, 2022-10-18:
Looks like both are bugs in meson.build.

infirit, 2022-10-19:
I also had to copy the systemd manager unit as it was missing. Now it works :balloon: . I also noticed that the applet systemd unit was missing.

I don't see where in meson copies the `org.blueman.Mechanism.conf` to `/etc/` though :thinking:

cschramm, 2022-10-19:
Double-checked the .deb files and they look good. I might have looked at the files of the installed package yesterday and it possibly inherited the etc path from the bullseye package. All good now, I guess.

closed 2022-10-19 by cschramm