blueman-project/blueman #1904: org.bluez.obex.Error.AlreadyExists Agent already exists

qaniqani, 2022-10-19:
blueman: 2.2.4
BlueZ: 5.64
Distribution: Ubuntu 22.04
Desktop environment: GNome

Hello,

There is no method left that I haven't researched and tried for a while.
But I'm using a Dell brand KM7321W 580 model keyboard and mouse on my Lenovo E14 G2 computer. When I want to connect the mouse via bluetooth, the connection cannot be made...

I completely deleted and reinstalled bluez, bluez tools and blueman-manager. I cleared the caches, still no results.

I noticed this in my transactions. Blueman-manager cannot find mouse in searches. When I open the settings -> bluetooth tab of the operating system, I can see the mouse, but when I say connect, it goes into off mode again.

I tried by marking the mac address of the device as trusted with Bluetoothctl from the terminal, it still didn't work.

It didn't happen on the command line, on the GUI.

In the blueman-applet logs it says:

ubuntu@Ubuntu-Pc:~$ blueman-applet --loglevel debug | grep EE:85
blueman-applet 16.41.47 INFO     PluginManager:85 load_plugin: ['TransferService', 'DhcpClient', 'GameControllerWakelock', 'Networking', 'DiscvManager', 'SerialManager', 'DBusService', 'RecentConns', 'AuthAgent', 'AppIndicator', 'KillSwitch', 'StatusIcon', 'NMPANSupport', 'ExitItem', 'AutoConnect', 'PPPSupport', 'ConnectionNotifier', 'NetUsage', 'ShowConnected', 'NMDUNSupport', 'Menu', 'StandardItems', 'DisconnectItems', 'PowerManager', 'Battery']
blueman-applet 16.41.47 ERROR    PluginManager:90 load_plugin: Unable to load plugin module GameControllerWakelock
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + f".{plugin}")
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/GameControllerWakelock.py", line 15, in <module>
    raise ImportError('This is not an X11 screen')
ImportError: This is not an X11 screen
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 16.41.47 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 16.41.47 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 16.41.47 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '10:3D:1C:D3:9C:6B', 'AddressType': 'public', 'Name': 'Ubuntu-Pc', 'Alias': 'Ubuntu-Pc', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 16.41.47 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 16.41.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 16.41.47 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 16.41.47 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 16.41.47 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 16.41.47 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 16.41.47 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.165
blueman-applet 16.41.47 INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.8
blueman-applet 16.41.47 INFO     Networking:71 set_nap   : set nap False
blueman-applet 16.41.47 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 16.41.47 INFO     RecentConns:107 initialize: rebuilding menu
blueman-applet 16.41.47 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 16.41.47 ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
blueman-applet 16.41.47 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 16.41.48 INFO     PowerManager:139 request_power_state: Requesting True
blueman-applet 16.41.48 INFO     KillSwitch:138 on_power_state_change_requested: True
blueman-applet 16.41.48 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 16.41.48 INFO     PowerManager:124 check     : callbacks done
blueman-applet 16.41.48 INFO     PowerManager:95 set_adapter_state: True
blueman-applet 16.41.48 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True

It would make me happy if you could help me on how to proceed.

Respects.

cschramm, 2022-10-19:
That message is definitely not the problem as OBEX is not even involved for such a device. You should check the logs of `bluetooth.service` when the connection fails. If it just reports timeouts reaching the device, it seems like the device does go offline indeed.

qaniqani, 2022-10-19:
> If it just reports timeouts reaching the device, it seems like the device does go offline indeed.

But it works directly on another windows device. It is unlikely that it is disabled, I use it actively.
I'm checking the logs of the Bluetooth services right now.
Thnx

qaniqani, 2022-10-19:
I have now re-examined the logs and found the following under the BUG heading:

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug.

Can anyone help from here?

Regards.

Oct 19 15:13:55 Ubuntu-Pc systemd[1]: Starting Bluetooth management mechanism...
Oct 19 15:13:55 Ubuntu-Pc dbus-daemon[751]: [system] Successfully activated service 'org.blueman.Mechanism'
Oct 19 15:13:55 Ubuntu-Pc systemd[1]: Started Bluetooth management mechanism.
Oct 19 15:13:56 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:14:19 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 36 times: [ unable to update icon for blueman]
Oct 19 15:14:19 Ubuntu-Pc firefox-esr.desktop[5998]: Unsupported modifier, resource creation failed.
Oct 19 15:14:19 Ubuntu-Pc firefox-esr.desktop[5998]: XXX: resource creation failed
Oct 19 15:14:20 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:14:24 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 9 times: [ unable to update icon for blueman]
Oct 19 15:14:26 Ubuntu-Pc systemd[1]: blueman-mechanism.service: Deactivated successfully.
Oct 19 15:14:29 Ubuntu-Pc systemd[2223]: app-gnome-gnome\x2dbluetooth\x2dpanel-5842.scope: Consumed 8.522s CPU time.
Oct 19 15:14:29 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:14:36 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 10 times: [ unable to update icon for blueman]
Oct 19 15:14:55 Ubuntu-Pc systemd[2223]: Started Application launched by gnome-shell.
Oct 19 15:14:55 Ubuntu-Pc gnome-bluetooth-panel.desktop[6879]: ../src/intel/isl/isl.c:2220: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
Oct 19 15:14:55 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:04 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 80 times: [ unable to update icon for blueman]
Oct 19 15:15:05 Ubuntu-Pc systemd[2223]: Started Application launched by gnome-shell.
Oct 19 15:15:05 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:05 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:05 Ubuntu-Pc gnome-shell[2539]: meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed
Oct 19 15:15:06 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:17 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 30 times: [ unable to update icon for blueman]
Oct 19 15:15:17 Ubuntu-Pc blueman-manager[5978]: blueman-manager 15.15.17 WARNING  ManagerDeviceMenu:141 fail      : fail g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusFailedError:  le-connection-abort-by-local
Oct 19 15:15:17 Ubuntu-Pc blueman-manager[5978]:  (0)
Oct 19 15:15:18 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:42 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 62 times: [ unable to update icon for blueman]
Oct 19 15:15:54 Ubuntu-Pc systemd[1]: Starting Load/Save RF Kill Switch Status...
Oct 19 15:15:54 Ubuntu-Pc systemd[1]: Started Load/Save RF Kill Switch Status.
Oct 19 15:15:54 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:58 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 24 times: [ unable to update icon for blueman]
Oct 19 15:15:58 Ubuntu-Pc blueman.desktop[3013]: blueman-applet 15.15.58 WARNING  PowerManager:202 on_adapter_property_changed: adapter powered on while in off state, turning bluetooth on
Oct 19 15:15:58 Ubuntu-Pc dbus-daemon[751]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.110' (uid=1000 pid=3013 comm="/usr/bin/python3 /usr/bin/blueman-applet ")
Oct 19 15:15:58 Ubuntu-Pc systemd[1]: Starting Bluetooth management mechanism...
Oct 19 15:15:58 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:15:58 Ubuntu-Pc dbus-daemon[751]: [system] Successfully activated service 'org.blueman.Mechanism'
Oct 19 15:15:58 Ubuntu-Pc systemd[1]: Started Bluetooth management mechanism.
Oct 19 15:15:58 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:16:03 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 40 times: [ unable to update icon for blueman]
Oct 19 15:16:03 Ubuntu-Pc systemd[1]: systemd-rfkill.service: Deactivated successfully.
Oct 19 15:16:03 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:16:25 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 33 times: [ unable to update icon for blueman]
Oct 19 15:16:28 Ubuntu-Pc systemd[1]: blueman-mechanism.service: Deactivated successfully.
Oct 19 15:16:29 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: unable to update icon for blueman
Oct 19 15:16:59 Ubuntu-Pc ubuntu-appindicators@ubuntu.com[2539]: message repeated 23 times: [ unable to update icon for blueman]
Oct 19 15:17:01 Ubuntu-Pc CRON[7032]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
[... cut ...]
Oct 19 15:24:53 Ubuntu-Pc dbus-daemon[751]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.153' (uid=1000 pid=7117 comm="/usr/bin/python3 /usr/bin/blueman-applet --logleve")
Oct 19 15:24:53 Ubuntu-Pc systemd[1]: Starting Bluetooth management mechanism...
Oct 19 15:24:53 Ubuntu-Pc dbus-daemon[751]: [system] Successfully activated service 'org.blueman.Mechanism'
Oct 19 15:24:53 Ubuntu-Pc systemd[1]: Started Bluetooth management mechanism.

qaniqani, 2022-10-19:
I think I got an idea of the problem. Does bluez provide support for products using BLE technology?
I've read multiple issues with this.

infirit, 2022-10-20:
Yes, BLE was implemented a long time ago. also those pixman issues are related to Cairo which is used by the toolkit Gtk+ we use.

The below snippet suggest you are having issues with you bluetooth adapter. 
```
Oct 19 15:17:05 Ubuntu-Pc kernel: [  608.224337] Bluetooth: hci0: unexpected cc 0x041a length: 7 > 1
Oct 19 15:17:05 Ubuntu-Pc kernel: [  608.224439] Bluetooth: hci0: Opcode 0x 41a failed: -22
```

And yes, this is not windows.

Are you using the device on windows, then on linux and back again?

qaniqani, 2022-10-22:
My mouse can also establish 3 different connections. It has 1 usb, 2 bluetooth channels.
It works fine on windows usb and 2nd bluetooth channel.
Ubuntu only sees the device on the 1st bluetooth channel, but reads the signals negatively.
I bought a bluetooth headset a few days ago and after a few tweaks in the bluez settings, I got the settings back. With these settings changes I had to reset the connected devices.
Before these processes, I was able to connect the mouse on the 1st bluetooth channel with a little forcing. But even though I reverted the changes, I can no longer connect.
This forces me to use usb dongle.

closed 2022-12-29 by github-actions[bot]