blueman-project/blueman #1905: blueman-manager items are "confused"

Harvie, 2022-10-19:
blueman: 2.3.2
BlueZ: 5.65
Distribution: ArchLinux
Desktop environment: XFCE

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

In blueman-manager there are multiple devices, but when i right click on some device and select eg. connect/disconnect. Different device gets (dis)connected rather than the one i've selected originaly. Perhaps there is some bug in the way the devices are reference internally?

infirit, 2022-10-19:
Can you run `blueman-manager --loglevel debug` from a terminal. Then connect/disconnect and post the complete log here.

drws, 2022-10-19:
I see this on Arch Linux which still has BlueMan v2.3.2 and I see some relevant changes in v2.3.3, so this might already be solved.

calvellido, 2022-10-26:
I just got a new computer, an HP ProBook 430 G8 with Linux Mint 21 installed:

```System:
  Kernel: 5.15.0-52-generic x86_64 bits: 64 compiler: gcc v: 11.2.0 Desktop: Cinnamon 5.4.12
    tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: HP product: HP ProBook 430 G8 Notebook PC v: N/A
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: HP model: 87DF v: KBC Version 33.19.00 serial: <superuser required> UEFI: HP
    v: T70 Ver. 01.08.02 date: 01/14/2022
Battery:
  ID-1: BAT0 charge: 33.8 Wh (98.0%) condition: 34.5/45.0 Wh (76.6%) volts: 12.4 min: 11.4
    model: Hewlett-Packard Primary serial: <filter> status: N/A
CPU:
  Info: quad core model: 11th Gen Intel Core i7-1165G7 bits: 64 type: MT MCP arch: Tiger Lake
    rev: 1 cache: L1: 320 KiB L2: 5 MiB L3: 12 MiB
  Speed (MHz): avg: 1186 high: 1567 min/max: 400/4700 cores: 1: 1244 2: 1303 3: 1232 4: 979
    5: 980 6: 1109 7: 1567 8: 1076 bogomips: 44851
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel TigerLake-LP GT2 [Iris Xe Graphics] vendor: Hewlett-Packard driver: i915
    v: kernel ports: active: HDMI-A-1,eDP-1 empty: DP-1,HDMI-A-2 bus-ID: 00:02.0 chip-ID: 8086:9a49
  Device-2: Chicony HP HD Camera type: USB driver: uvcvideo bus-ID: 3-2:3 chip-ID: 04f2:b6c0
  Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: modesetting unloaded: fbdev,vesa
    gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 5360x1440 s-dpi: 96
  Monitor-1: HDMI-1 mapped: HDMI-A-1 pos: right model: BenQ EX3501R res: 3440x1440 dpi: 107
    diag: 889mm (35")
  Monitor-2: eDP-1 pos: primary,left model: Chi Mei Innolux res: 1920x1080 dpi: 166
    diag: 336mm (13.2")
  OpenGL: renderer: Mesa Intel Xe Graphics (TGL GT2) v: 4.6 Mesa 22.0.5 direct render: Yes
Audio:
  Device-1: Intel Tiger Lake-LP Smart Sound Audio vendor: Hewlett-Packard
    driver: sof-audio-pci-intel-tgl bus-ID: 00:1f.3 chip-ID: 8086:a0c8
  Sound Server-1: ALSA v: k5.15.0-52-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Wi-Fi 6 AX201 driver: iwlwifi v: kernel bus-ID: 00:14.3 chip-ID: 8086:a0f0
  IF: wlp0s20f3 state: down mac: <filter>
  Device-2: Realtek RTL8153 Gigabit Ethernet Adapter type: USB driver: r8152 bus-ID: 2-1.4:5
    chip-ID: 0bda:8153
  IF: enx00e14c68056b state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Intel AX201 Bluetooth type: USB driver: btusb v: 0.8 bus-ID: 3-10:7 chip-ID: 8087:0026
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 3.0 lmp-v: 5.2
    sub-v: 237e
RAID:
  Hardware-1: Intel Volume Management Device NVMe RAID Controller driver: vmd v: 0.6
    bus-ID: 00:0e.0 chip-ID: 8086:9a0b
Drives:
  Local Storage: total: 476.94 GiB used: 22.41 GiB (4.7%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: PC SN530 SDBPNPZ-512G-1006 size: 476.94 GiB
    speed: 31.6 Gb/s lanes: 4 serial: <filter> temp: 36.9 C
Partition:
  ID-1: / size: 284.03 GiB used: 22.29 GiB (7.8%) fs: ext4 dev: /dev/nvme0n1p5
  ID-2: /boot/efi size: 256 MiB used: 128.4 MiB (50.1%) fs: vfat dev: /dev/nvme0n1p1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
Sensors:
  System Temperatures: cpu: 49.0 C mobo: 25.0 C
  Fan Speeds (RPM): N/A
Repos:
  Packages: apt: 2225
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/caldas-lopes-ppa-jammy.list
    1: deb http: //ppa.launchpad.net/caldas-lopes/ppa/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/home:manuelschneid3r.list
    1: deb http: //download.opensuse.org/repositories/home:/manuelschneid3r/xUbuntu_22.04/ /
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
[... 775 lines cut ...]
blueman-manager 12.26.35 DEBUG    PulseAudioProfile:46 on_pa_event: 25 1
blueman-manager 12.26.35 INFO     PulseAudioProfile:58 on_pa_event: card
blueman-manager 12.26.35 INFO     PulseAudioProfile:60 on_pa_event: change
blueman-manager 12.26.35 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
blueman-manager 12.26.40 INFO     PulseAudioUtils:324 __event_callback: 25 1
blueman-manager 12.26.40 DEBUG    PulseAudioProfile:46 on_pa_event: 25 1
blueman-manager 12.26.40 INFO     PulseAudioProfile:58 on_pa_event: card
blueman-manager 12.26.40 INFO     PulseAudioProfile:60 on_pa_event: change
blueman-manager 12.26.40 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
blueman-manager 12.26.48 INFO     PulseAudioUtils:324 __event_callback: 25 1
blueman-manager 12.26.48 DEBUG    PulseAudioProfile:46 on_pa_event: 25 1
blueman-manager 12.26.48 INFO     PulseAudioProfile:58 on_pa_event: card
blueman-manager 12.26.48 INFO     PulseAudioProfile:60 on_pa_event: change
blueman-manager 12.26.48 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
blueman-manager 12.27.01 INFO     PulseAudioUtils:324 __event_callback: 25 1
blueman-manager 12.27.01 DEBUG    PulseAudioProfile:46 on_pa_event: 25 1
blueman-manager 12.27.01 INFO     PulseAudioProfile:58 on_pa_event: card
blueman-manager 12.27.01 INFO     PulseAudioProfile:60 on_pa_event: change
blueman-manager 12.27.01 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
blueman-manager 12.27.08 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
blueman-manager 12.27.12 INFO     ManagerDeviceList:427 row_update_event: Trusted True
blueman-manager 12.27.12 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
blueman-manager 12.27.12 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_00_16_94_2D_EA_35 {'Trusted': True}
blueman-manager 12.27.13 DEBUG    ManagerDeviceMenu:299 generate  : PXC 550
```

tabcodes, 2022-10-30:
For whatever it's worth- I'm also suffering this after upgrading to Mint 21 (which also upgraded `blueman` to `2.3.4+mint1+vanessa`). It seems that right-clicking any Bluetooth device entry and looking at the `Info` will yield a different device's information- but doing the same from the `Device > Info` drop-down at the top yields the correct info.

Downgrading to `blueman 2.2.4` solved the problem for me, but YMMV.

infirit, 2022-10-31:
I think I know what is going on. Can you disable hiding unnamed devices in the view menu.

I can reproduce but I don't have many unnamed devices close to me.

infirit, 2022-10-31:
#1918 should fix this.

closed 2022-11-10 by cschramm