blueman-project/blueman #1907: Paired Bluetooth headsets won't connect

josh-bartz, 2022-10-19:
blueman: 2.3.3+mint1+vanessa
BlueZ: 5.6.4-0ubuntu1
Distribution: Linux Mint 21
Desktop environment: Cinnamon 5.4.12

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

kvelakur, 2022-10-19:
This is a [bug](https://github.com/blueman-project/blueman/issues/1887) on blueman 2.3.3 which is the version available on the Linux Mint 21 main repository. So all PCs running LM21 have broken a broken bluetooth manager at the moment. 

Do one of the maintainers know when this bugfix will be released, and when we can expect it to appear on the main repository?

infirit, 2022-10-19:
2.3.4 was release specifically to address this. https://github.com/blueman-project/blueman/releases/tag/2.3.4

closed 2022-10-19 by cschramm