blueman-project/blueman #1909: No input device for my BT headphones

birdie-github, 2022-10-20:
blueman:  blueman-2.2.5-1.fc36.x86_64
BlueZ: bluez-5.65-1.fc36.x86_64
Distribution: Fedora 36 fully updated sans blueman (don't like the new icons)
Desktop environment: XFCE

I don't see my BT headphones as an input device though they absolutely have this profile and must be able to at least send PAUSE/PLAY events.

Volume up/down buttons do work.

In `/etc/bluetooth/main.conf` everything is by default except `Experimental = true` and `KernelExperimental = true`.

`xinput test-xi2 --root` reports no events from the headphones.

Please advise.

```
$ xinput 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech G203 Prodigy Gaming Mouse      	id=11	[slave  pointer  (2)]
⎜   ↳ Logitech G203 Prodigy Gaming Mouse Keyboard	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard                   	id=8	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard Consumer Control  	id=9	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard System Control    	id=10	[slave  keyboard (3)]
    ↳ Logitech G203 Prodigy Gaming Mouse Keyboard	id=13	[slave  keyboard (3)]
```

```
public
Bluedio T
Bluedio T
0x240404
0x0000
audio-headset
yes
yes
no
no
yes
0000110b-0000-1000-8000-00805f9b34fb Audio Sink
0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
0000110d-0000-1000-8000-00805f9b34fb Advanced Audio
0000110e-0000-1000-8000-00805f9b34fb Remote Control
0000111e-0000-1000-8000-00805f9b34fb Handsfree
/org/bluez/hci0
```

[image]

cschramm, 2022-10-20:
Running `bluetoothd` with `-d profiles/audio/avctp.c` gives me the following when connecting an AVRCP device:

```
bluetoothd[6616]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connecting:
bluetoothd[6616]: /org/bluez/hci0/dev_A4_77_58_00_11_8E/sep5/fd0: fd(42) ready
bluetoothd[6616]: profiles/audio/avctp.c:avctp_connect_cb() AVCTP: connected to A4:77:58:00:11:8E
bluetoothd[6616]: profiles/audio/avctp.c:init_uinput() AVRCP: uinput initialized for SHIVR NC18
bluetoothd[6616]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connected
```

birdie-github, 2022-10-20:
```
bluetoothd[942789]: profiles/audio/avctp.c:uinput_create() Can't open input device: No such file or directory (2)
bluetoothd[942789]: profiles/audio/avctp.c:init_uinput() AVRCP: failed to init uinput for Bluedio T
```

closed 2022-10-20 by birdie-github

birdie-github, 2022-10-20:
@cschramm 

My custom kernel didn't support/include the necessary option, # CONFIG_INPUT_UINPUT is not set.

I've just enabled it and everything works.

Maybe you could check for the existence of this kernel module/interface and give a more human-readable error in this case.

Sorry for the noise!

cschramm, 2022-10-20:
This is far off blueman's scope. There was not any error message in blueman, if I got you right, and none is expected as the connection does work in general.

birdie-github, 2022-10-20:
> This is far off blueman's scope. There was not any error message in blueman, if I got you right, and none is expected as the connection does work in general.

You're right, it's for bluez, I just thought you're a developer behind both projects.

infirit, 2022-10-20:
> I just thought you're a developer behind both projects.

Yeah, we get that a lot unfortunately. People think the frontend Blueman is responsible for everything bleutooth on their system. Something we will have to live with I guess.