blueman-project/blueman #1910: how disable autostart in manjaro-xfce

Phebo90, 2022-10-20:
Hi all, 
i have a problem with autostart with this tool. When i login in OS it opens up on desktop and i have to close it manually.
I tried deflag in  Session & Startup, i tried at remove /etc/xdg/autostart/blueman.desktop but when i login again, this reopens

cschramm, 2022-10-20:
Please elaborate what opens up. `blueman.desktop` only starts `blueman-applet` which is a background process.

cschramm, 2022-10-20:
That is `blueman-manager`. I'm not aware of anything that autostarts it.

Do you save your session? Checking the parent processes (e.g. `ps -e f`, `pstree`, ...) could give you a hint what's starting it.

Phebo90, 2022-10-20:
i have only this reference inherent in blueman:

        ├─systemd─┬─(sd-pam)
        │         ├─at-spi-bus-laun─┬─dbus-daemon
        │         │                 └─3*[{at-spi-bus-laun}]
        │         ├─at-spi2-registr───2*[{at-spi2-registr}]
        │         ├─blueman-applet───3*[{blueman-applet}]
        │         ├─blueman-tray───2*[{blueman-tray}]

cschramm, 2022-10-20:
There must be a `blueman-manager` process as well as that's its window.

Phebo90, 2022-10-20:
i found this in journalctl:

Oct 20 23:55:38 XXX-20h5cto1ww dbus-daemon[1871]: [session uid=1000 pid=1871] Activating via systemd: service name='org.blueman.Applet' unit='blueman-applet.service' requested by ':1.55' (uid=1000 pid=2149 comm="/usr/bin/python /usr/bin/blueman-manager")

cschramm, 2022-10-20:
When you see that, you can check the sender (`:1.55` in your case, but it's temporary) e.g. in `d-feet` (Session bus!). Is it the `blueman-applet` process?

Phebo90, 2022-10-22:
> When you see that, you can check the sender (`:1.55` in your case, but it's temporary) e.g. in `d-feet` (Session bus!). Is it the `blueman-applet` process?

I don't found nothing

cschramm, 2022-10-22:
You see that D-Bus activation took place but cannot find the process that requested it on the bus? :thinking: That's weird. It would have been a process that's not running anymore then. Would be hard to catch without any trace. Did you try disabling all kinds of things in your autostart settings and see if it still happens?

infirit, 2022-10-22:
You should also check your saved session in xfce.

Phebo90, 2022-10-25:
Hi all,
after many attempts at trouble shooting i found the problem, i removed a saved session and after reboot now don't spawn blue-manager. 
I have no idea how this is possible but i resolved.

Thx all

closed 2022-10-25 by cschramm