blueman-project/blueman #1912: Playstation DualShock 4 Error: Host is down

Exagram, 2022-10-22:
blueman: blueman-manager 2.1.2
BlueZ: ???
Distribution: Linux Mint 20.3
Desktop environment: Cinnamin 64-bit

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

- Context: A few months I was able to use my joystick with Steam games (Elden Ring). All I did was uninstall the game.

Steps:
- Have Steam running
- Enabled bluetooth
- Open Blueman > Devices > Device > Connect
- Turn on my joystick
- For a brief second I see it connected
- After 1 second I see a red bar with "Error: Host is down"

Diagnostics:
- `blueman-applet --loglevel debug`: https://gist.github.com/Exagram/ef7082052a47fbbeff87c4e90b5b2837
- `cat /var/log/syslog`: https://gist.github.com/Exagram/cc864a3bd8850915975d48dd5cbe876a

cschramm, 2022-10-22:
Looks like some BlueZ state issue to me.

The device does not seem to be known to BlueZ, so it's not paired. Did you try pairing?

(Ubuntu 20.04 has BlueZ 5.53, so Linux Mint 20.3 has the same version)

Exagram, 2022-10-24:
I tried pairing but after a few seconds the Dualshock Controller entry disappears from the list.

I have removed the entry and repaired/reconfigured it but no luck.

I have removed all plugins and still no luck.

closed 2022-12-31 by github-actions[bot]