blueman-project/blueman #1916: Beyerdynamics Headphones do not work with my laptop but they work with my Desktop PC

hiphopconnect, 2022-10-26:
Hello all,

I am new here. I hope I'm doing everything right, I'm quite new to all this. I hope you can help me. 

****Desktop PC:**** Ear-enclosing headphones work. In-ear headphones work.

Blueman: 2.12
BlueZ: 5.53
Distribution:
Desktop environment:
`hiphopconnect@EMC22:~$ cat /etc/os-release
NAME="Linux Mint"
VERSION="20.3 (Una)"
UBUNTU_CODENAME=focal
No LSB modules are available.
5.4.0-126-generic
`

**Laptop:** Ear-enclosing headphones do NOT work. In-ear headphones work.

Blueman: 2.3.4
BlueZ: 5.64
Distribution: 
Desktop environment:
`
NAME="Linux Mint"
VERSION="21 (Vanessa)"
UBUNTU_CODENAME=jammy
No LSB modules are available.
Kernel: Linux 5.15.0-52-generic
Architecture: x86-64
Hardware Vendor: Lenovo
Hardware Model: ThinkPad L15 Gen 2a
`

Unfortunately, my ear-enclosing Bluetooth headphones (headset) don't work on my laptop and I've tried everything. I hope you can help me.

With these instructions ([Bluetooth Probleme lösen](https://www.pcwelt.de/article/1177607/typische-bluetooth-probleme-in-linux-loesen.html)), I was able to integrate the headphones wonderfully under my desktop computer. Attached are the specifications.

Unfortunately it doesn't work on my laptop. Please find attached the specifications.

I just don't know what to do. I don't understand what the difference is. Attached are the data sheets of the two headphones. The in-ear headphones (headset) work perfectly. Unfortunately, the ear-enclosing headphones don't work at all. [Ear-closed you must scroll down](https://www.beyerdynamic.de/lagoon-anc-explorer.html) [In-Ear ](https://www.beyerdynamic.de/blue-byrd.html)

****Desktop PC:**** Ear-enclosing headphones work. In-ear headphones work.

****Laptop:**** Ear-enclosing headphones do NOT work. In-ear headphones work.
`
hiphopconnect@EMC25:~$ blueman-applet --loglevel debug
blueman-applet 19.20.26 INFO     PluginManager:87 load_plugin: ['DisconnectItems', 'AutoConnect', 'ConnectionNotifier', 'SerialManager', 'DiscvManager', 'GameControllerWakelock', 'RecentConns', 'Menu', 'ShowConnected', 'DBusService', 'KillSwitch', 'PowerManager', 'DhcpClient', 'NMDUNSupport', 'PPPSupport', 'ExitItem', 'NetUsage', 'NMPANSupport', 'StandardItems', 'AuthAgent', 'StatusNotifierItem', 'StatusIcon', 'TransferService', 'Networking']
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 19.20.26 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'E0:0A:F6:65:B5:40', 'AddressType': 'public', 'Name': 'EMC25', 'Alias': 'EMC25', 'Class': 8126732, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 19.20.26 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
[... 75 lines cut ...]
blueman-applet 19.22.28 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 19.22.28 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.71
blueman-applet 19.22.28 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.232
blueman-applet 19.22.28 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_00_22_BB_76_0D_E4 {'Address': '00:22:BB:76:0D:E4', 'AddressType': 'public', 'Name': 'Blue Byrd (2nd gen.)', 'Alias': 'Blue Byrd (2nd gen.)', 'Class': 2360344, 'Icon': 'audio-headphones', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00000000-deca-fade-deca-deafdecacaff', '00001101-0000-1000-8000-00805f9b34fb', '00001107-d102-11e1-9b23-00025b00a5a5', '00001108-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '931c7e8a-540f-4686-b798-e8df0a2ad9f7'], 'Modalias': 'usb:v2E50p000Bd0107', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
blueman-applet 19.22.28 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_00_22_BB_74_14_91 {'Address': '00:22:BB:74:14:91', 'AddressType': 'public', 'Name': 'Lagoon ANC ', 'Alias': 'Lagoon ANC ', 'Class': 2360344, 'Icon': 'audio-headphones', 'Paired': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000110d-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000110f-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '00001131-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
blueman-applet 19.22.28 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 19.22.28 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 19.22.28 INFO     Networking:71 set_nap   : set nap False
blueman-applet 19.22.28 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 19.22.28 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
[... cut ...]
blueman-applet 19.22.29 INFO     KillSwitch:138 on_power_state_change_requested: True
blueman-applet 19.22.29 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 19.22.29 INFO     PowerManager:125 check     : callbacks done
blueman-applet 19.22.29 INFO     PowerManager:96 set_adapter_state: True
blueman-applet 19.22.29 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
`

If you need more information, let me know. 

Thanks for your help.

Michael

cschramm, 2022-10-27:
Please elaborate on "do not work".

Obviously different hardware and thus drivers are involved, but also the BlueZ and pulseaudio versions differ significantly.

hiphopconnect, 2022-10-28:
I would be happy to explain to you what does not work. After a successful connection, the connection is immediately disconnected again, the disconnection takes place immediately after the connection. 

I can't even change the audio profile as I have to do under the desktop version in order to use the headphones as a headset.

cschramm, 2022-11-22:
Not sure what logs you posted but the `Lagoon ANC` do not show up as paired. Did you even successfully pair them?

hiphopconnect, 2022-11-23:
Yes of course. For a moment. After a few seconds the connections break up. I paired it successfully in the command line. Hope you can help me. 
Michael from Augsburg

cschramm, 2022-11-23:
I'd check bluetoothd, kernel and pulseaudio logs. In case there's nothing, bluetoothd debug logging should uncover the disconnect reason. The fix might just be something like resetting and / or re-pairing the device (potentially re-reading its manual).

hiphopconnect, 2023-02-09:
Hello, unfortunately the headphones still don't work with either my desktop PC or my laptop. They can be connected with Bluetooth, but they are not recognised as an input and output source and I can no longer adjust their profile. I would be happy if you had a tip on how to make them known to the system as such. Many thanks in advance. Best regards Michael from Augsburg

Hallo leider funktionieren die Kopfhörer immer noch nicht weder mit meinem Desktop PC noch mit meinem Laptop. Sie lassen sich mit dem Bluetooth verbinden, Allerdings werden sie nicht als Ein- und Ausgangsquelle erkannt und ihr Profil kann ich auch nicht mehr anpassen. Ich würde mich freuen wenn sie einen Tipp hätten wie man sie als solche dem System kenntlich macht. Vielen Dank im voraus. Beste Grüße Michael aus Augsburg

Laptop 
System:
  Kernel: 5.15.0-58-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.7
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: LENOVO product: 20X7003TGE v: ThinkPad L15 Gen 2a
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: LENOVO model: 20X7003TGE serial: <superuser required> UEFI: LENOVO v: R1KET39W (1.24 )
    date: 08/19/2022
CPU:
  Info: 8-core model: AMD Ryzen 7 PRO 5850U with Radeon Graphics bits: 64 type: MT MCP arch: Zen 3
    rev: 0 cache: L1: 512 KiB L2: 4 MiB L3: 16 MiB
Audio:
  Device-1: AMD Renoir Radeon High Definition Audio vendor: Lenovo driver: snd_hda_intel v: kernel
    pcie: speed: 8 GT/s lanes: 16 bus-ID: 07:00.1 chip-ID: 1002:1637
  Device-2: AMD Raven/Raven2/FireFlight/Renoir Audio Processor vendor: Lenovo driver: N/A pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 07:00.5 chip-ID: 1022:15e2
  Device-3: AMD Family 17h HD Audio vendor: Lenovo driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 07:00.6 chip-ID: 1022:15e3
  Device-4: Beyerdynamic FOX type: USB driver: hid-generic,snd-usb-audio,usbhid bus-ID: 3-2:5
    chip-ID: 2e50:000e
  Sound Server-1: ALSA v: k5.15.0-58-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Bluetooth:
  Device-1: Realtek Bluetooth Radio type: USB driver: btusb v: 0.8 bus-ID: 5-4:2
    chip-ID: 0bda:4852
  Report: hciconfig ID: hci0 rfk-id: 4 state: up address: <filter> bt-v: 3.0 lmp-v: 5.2
    sub-v: 6d7a