blueman-project/blueman #1917: Connecting and disconnecting intermittently, not working properly when connected

512marek, 2022-10-30:
blueman: 19.19.58
BlueZ: 5.64
Distribution: Linux Mint 21 Cinnamon
Desktop environment: Cinnamon 5.4.12

https://gist.github.com/512marek/441c519dd0acc2a4703945c5794858e8

Hello,
I was trying to connect to my earbuds ('Name': 'Mi True Wireless EBs Basic_R' in the logs), and after succeeding the system started to disconnect and connect to them intermittently (10 events in total, first at 18.49.20 and the last at 19.02.09 according to Cinnamon notifications.). Furthermore, during the connection Tidal (opened in the browser) was playing the music intermittently. I know that everything is okay with the headphones, as I use them on the daily basis. Bug seems to be reproducible (even after restarting etc.) and as I recall it also happened with other devices (though not so frequently)
Also when checking blueman version ("terminal output" in gists) with 'blueman-applet' some warnings appeared.

In addition I have to enable bluetooth with that command, otherwise bluetooth manager page is blank and nothing can be done:
sudo modprobe -r btusb && sleep 20 && sudo modprobe btusb
(It is a separate problem but I wanted to mentioned it)

I will be happy to provide any further details

cschramm, 2022-11-22:
The Internets have https://www.reddit.com/r/archlinux/comments/sh6eqs/bluetooth_headset_disconnecting_intermittently/ which seems to match pretty good. tl;dr: Broken firmware on Intel AX200 / AX201 controllers.

512marek, 2022-12-03:
Some recent update fixed that connection interruption, so the reddit thread was right, it was something with the drivers.

closed 2022-12-03 by 512marek