blueman-project/blueman #1921: Window that asks for conformation code for pairing is not shown

Dmitry-Borodin, 2022-11-05:
blueman: 2,3,4+mint1+vanessa
BlueZ: 5.64-0-ubuntu-1
Distribution: Mint 21
Desktop environment: Cinnamon

I have Logitec bt keyboard that asks to type passcode while pairing to pair. I used Blueman ~ a year ago with this keyboard and it was working.
Now I see only error overlay after timeout "pairing failed"

Had to use  bluetoothctl command line tool to successfully pair - it shows passcode as it should.

infirit, 2022-11-08:
Post the blueman-applet log when you are trying to pair the keyboard. It's describes in the troubleshooting page you were linked to in the issue templaye

Dmitry-Borodin, 2022-11-08:
Thx,
I will do that next time I'll need to connect this keyboard - afraid if I"ll unpair it would be difficult to pair again -)

closed 2022-11-08 by Dmitry-Borodin