blueman-project/blueman #1922: Playstation Dualshock 3 Connected, but can't pair

Handika997, 2022-11-06:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2022-11-08:
It may be because you don't have the bluez-plugin installed. I do have an original one i can try later but everything you need is in BlueZ and the kernel these days so no need for additional things like qtsixa

infirit, 2022-11-08:
So the process

1. Connect the ps3 controller with a cable
2. Accept the authentication (always accept)
3. Done
4. Unplug

Handika997, 2022-11-09:
> It may be because you don't have the bluez-plugin installed. I do have an original one i can try later but everything you need is in BlueZ and the kernel these days so no need for additional things like qtsixa

can you kindly tell me the steps please

infirit, 2022-11-09:
The sixaxis plugin is location in the bluez-plugin directory, for me this is `/usr/lib/bluetooth/plugins/sixaxis.so`. The steps how to pair are already provied.

closed 2023-01-16 by github-actions[bot]