blueman-project/blueman #1925: cython and python 3.11

infirit, 2022-11-11:
Gentoo has ways to deal with this but just want to put this out here that this is an issue.

tnut, 2022-11-20:
Any news on this issue? 
I got the same error on my testing branch of NuTyX using python 3.11

infirit, 2022-11-20:
> Any news on this issue? I got the same error on my testing branch of NuTyX using python 3.11

Just remove the `_blueman.c` before building and it will get generated again with the newer cython version.

infirit, 2022-11-20:
Also, to be clear, this is not a bug. Unfortunate yes. You should always regenerate, it's also what Gentoo recommends in their [developer documentation](https://projects.gentoo.org/python/guide/distutils.html#packages-using-cython).

tnut, 2022-11-21:
Thanks a lot for you quick response. [solved]

cschramm, 2022-11-21:
What are the implications here? Is it a problem with meson? With autotools it just means that we should not ship `_blueman.c` in the tarballs, right?

infirit, 2022-11-21:
In automake we can't really get around it as it in BUILD_SOURCES. Meson should be unaffected. It's one of those unfortunate things distros have to deal with.

infirit, 2022-11-21:
Sorry, I was unclear. automake needs `_blueman.c` as it's required for building the module. It doesn't know it's a generated file and includes it in the tarball. 

To make sure it's generated before the module is build it's added to `BUILT_SOURCES` which forces cython to run on make all/dist/check and a couple more.

cschramm, 2022-11-21:
:ok_hand: We could still just remove it from the tarball, no? Anyway, if we drop autotools, it's not a problem anymore.

tnut, 2022-11-22:
For info, I still build via "./configure && make && make install" without  issues. So yes you can remove it. When you are able to generate files source code files   at "compile time" it's by far the best option.

infirit, 2022-11-22:
> We could still just remove it from the tarball, no?

Maybe, let me look if automake can do this.

infirit, 2022-11-22:
Should be good in 2.3.5.

closed 2022-11-22 by infirit