blueman-project/blueman #1929: Pairing request for <device>: endless loop

doebi, 2022-11-15:
blueman: 16.20.27
bluetoothctl: 5.49
Distribution: Ubuntu 22.04
Desktop environment: i3

* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I got myself a brand new bluetooth mouse and wanted to get going, but instead I ended up here...

First off, scan and discovery works great in blueman gui. 
Connecting to the mouse also works, but then it gets tricky.
Right after the mouse is connected, it sends a pairing request. I get the notification, but when I click on it, nothing happens. And when issuing a pairing request from my side it also fails. I tried setting the device as "trusted" in the hope for a pairing request to get auto-accepted, but without any change.

[image]

So now I am in an endless loop where the mouse repeatedly connected, sends a pairing request and disconnects to start over again.

How can make it work?, Thanks.

infirit, 2022-11-16:
The pairing never properly succeeded so blueman will ask you again when the device tries to connect. The notification has action(d) which you will need to perform which is guess you aren't doing. In the screenshot it looks like it's the letter a.

doebi, 2022-11-16:
Thanks for your message. So what does this mean for me? How can make it work?, Thanks.

infirit, 2022-11-16:
Like I said perform the action that the notification shows you. Here is how dunst handles these actions https://dunst-project.org/documentation/#ACTIONS

doebi, 2022-11-16:
I replaced dunst with rofi, which shows a proper action dialoge and lets me complete the pairing.

closed 2022-11-16 by doebi