blueman-project/blueman #1933: No module named '_blueman'

BambValley, 2022-11-17:
blueman: 2.0.5
BlueZ:
Distribution: ubuntu18.04
Desktop environment:

i download blueman with "sudo apt-get install blueman",  and run with "./blueman-manager" in "/usr/bin".
however, i got errors as followed:

zhang@zhang:/usr/bin$ ./blueman-manager 
Traceback (most recent call last):
  File "./blueman-manager", line 26, in <module>
    from blueman.main.Device import Device
  File "/usr/lib/python3/dist-packages/blueman/main/Device.py", line 14, in <module>
    import blueman.services
  File "/usr/lib/python3/dist-packages/blueman/services/__init__.py", line 8, in <module>
    from blueman.services.DialupNetwork import DialupNetwork
  File "/usr/lib/python3/dist-packages/blueman/services/DialupNetwork.py", line 6, in <module>
    from blueman.services.meta import SerialService
  File "/usr/lib/python3/dist-packages/blueman/services/meta/__init__.py", line 7, in <module>
    from blueman.services.meta.SerialService import SerialService
  File "/usr/lib/python3/dist-packages/blueman/services/meta/SerialService.py", line 7, in <module>
    from _blueman import rfcomm_list, release_rfcomm_device, create_rfcomm_device
ModuleNotFoundError: No module named '_blueman'

i checked the path "/usr/lib/python3/dist-packages/blueman/services/meta/", and found there was indeed no file named "_blueman.py".
and now i dont know how use blueman

infirit, 2022-11-17:
_blueman is a binary module with a full path of `/usr/lib/python3/dist-packages/_blueman.cpython-36m-x86_64-linux-gnu.so` is it there?

BambValley, 2022-11-17:
> _blueman is a binary module with a full path of `/usr/lib/python3/dist-packages/_blueman.cpython-36m-x86_64-linux-gnu.so` is it there?

thx for your reply.
yes, that binary file is there.i noticed the python in my pc is 3.8, so i made another copy named "_blueman.cpython-38m-x86_64-linux-gnu.so". Sadly, that makes no difference.

BambValley, 2022-11-18:
> _blueman is a binary module with a full path of `/usr/lib/python3/dist-packages/_blueman.cpython-36m-x86_64-linux-gnu.so` is it there?

I've got this work.
Thanks for your reminding. I realize that python 3.8 does not match blueman 2.0.5.
I downgrade python to 3.6 and blueman work normally. 
Thanks a lot.

closed 2022-11-18 by infirit