blueman-project/blueman #1938: Sony headphone WH-1000XM5 does not work

flickleafy, 2022-11-20:
blueman: 2.2.4
BlueZ: 5.64
Distribution: Ubuntu 22.04
Desktop environment: GNOME 42.0

I paired this headphone, and tried to use, but nothing works. I have no idea what is actually happening. I only know it is not recognized as audio device by blueman.

cschramm, 2022-11-22:
From a quick Internet search I'm pretty sure you do not want to use the `LE_...` device but its non-`LE_...` pendant that you can find when in pairing mode.

flickleafy, 2022-11-22:
I am not sure what you mean, there is not any "non-LE_..." device.... can you explain better?

cschramm, 2022-11-22:
See e.g. https://askubuntu.com/questions/1393926/sony-wf-l900-linkbuds-bluetooth-earbuds-not-recognized-as-audio-device. I guess if you discover only `LE_WH-1000XM5` but no `WH-1000XM5`, the device is not in pairing mode. Connecting to `LE_WH-1000XM5` seems pointless as it's just a collection of proprietary services (probably for setup and controlling) but no audio services.

flickleafy, 2022-11-23:
I highly doubt it is similar to that "linkbuds" model, since that does need the case do pair, and that probably was the user mistake. I can pair my headphone in a laptop with windows, and in android without any issue, there is just one way to put it to pairing mode, hold the button for 5 seconds, and there is no other way.
https://helpguide.sony.net/mdr/wh1000xm5/v1/en/contents/TP1000534541.html

Also, it is worth mentioning that I have a couple other bluetooth devices that fail to pair in Ubuntu (desktop), and they also have that prefix "LE-" at the beginning of the device name. 

Is there any chance the problem is the bluetooth adapter? It is a combo Wifi6E+Bluetooth, from a new desktop motherboard I bought. The vendor is MediaTek, model MT7922

cschramm, 2022-11-23:
> I can pair my headphone in a laptop with windows, and in android without any issue

Is it connected as `LE_WH-1000XM5` there as well?

> Is there any chance the problem is the bluetooth adapter?

Not really. The connected device just does not offer any audio or other useful service. I don't see how the adapter could trigger that.

kuraga, 2022-12-17:
Get `le-connection-abort-by-local` on connection.

Also, sometimes I get even error from #895 instead:
```
blueman.bluez.errors.BluezDBusException: org.freedesktop.DBus.Error.UnknownObject Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

kuraga, 2022-12-19:
BTW, is it Blueman or BlueZ issue?

cschramm, 2022-12-19:
@kuraga: I fail to fit your comments to this issue.

kuraga, 2022-12-19:
@cschramm , first comment is about Sony WH-1000XM5 headphone connection.

infirit, 2022-12-19:
What happens when both of you set the adapter visible via blueman-adapters?

flickleafy, 2022-12-24:
I managed to pair my headphone and now it shows as it should be, but, I got another type of error... 

"Connection Failed: br-connection-profile-unavailable"

infirit, 2022-12-24:
You are most likely missing the pulseaudio Bluetooth module (pulseaudio-module-bluetooth).

flickleafy, 2022-12-24:
I did this command on console... and got this response...

`sudo apt-get -y install pulseaudio-module-bluetooth
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
pulseaudio-module-bluetooth is already the newest version (1:15.99.1+dfsg1-1ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

flickleafy, 2022-12-24:
I did a new debug, starting from zero, pairing, then connecting, and after it connected, it gave the error, and removed itself from the device list...

[](https://gist.github.com/flickleafy/d538074b1fae6502018074df24ad9c91)

cschramm, 2022-12-24:
I would clearly not expect it in your setting but you might be using PipeWire. In that case you need to install libspa-0.2-bluetooth and stop / restart all PipeWire-related services.

Any log messages from bluetooth.service or pulseaudio.service?

flickleafy, 2022-12-24:
That worked! Thank you

I just noticed that PipeWire is installed here, and that is probably a better option, since PulseAudio does not work with the newer codecs, and the newer codecs are available in PipeWire.

I searched for the libspa-0.2-bluetooth, and I reached this site (PipeWire section), that also helped me with some other details in the configuration:
https://wiki.debian.org/BluetoothUser/a2dp

closed 2022-12-24 by flickleafy

flickleafy, 2022-12-25:
Also, one other detail that I found quite annoying, was that in my current OS (Ubuntu), it does not re-connect to the device, when it comes online, even if it marked as trusted, and I dont know the reason, then I just found a script that will help with the re-connection if for any reason the device goes offline, and you get it online again...

The script on step 5 will solve the issue:
https://techwiser.com/fix-bluetooth-device-doesnt-auto-connect-in-linux/

infirit, 2022-12-25:
> it does not re-connect to the device, when it comes online, even if it marked as trusted

This is what blueman's autoconnect plug-in is for. It's in the right-click devicemenu.

kuraga, 2022-12-26:
@flickleafy , so, does your headphones work in both A2DP and HSP/HFP profiles?

flickleafy, 2022-12-26:
Honestly, I still had issues with blueman some times, it pop up that connection error message sporadically.

I thought it was some kind of issue with blueman installation, then, I uninstalled and purged all its configurations.

And also, I installed the autoconnect script on the link above.

Now it works as expected... look:
[image]

cschramm, 2022-12-27:
To avoid confusion: Ubuntu 22.04's pulseaudio does support LDAC and AptX and blueman >=2.2 ships an auto-connect toggle for supported services (including audio) in the device's right-click menu.

flickleafy, 2022-12-29:
Yes, you are right @cschramm , the new pulseaudio 15 support that codecs, but, to give some context, actually my computer had Ubuntu 20.04 installed, and at that time, pulseaudio had many issues, involving not only the lack of some codecs, but also latency issues, and at that time, I installed PipeWire.

So 1 year (ish) has passed, and I upgraded my Ubuntu to 22.04, and I forgot I had installed PipeWire.

About the autoconnect, I knew it was available on blueman, but, that is not the reason I uninstalled blueman, and used the autoconnect script that I mentioned earlier. I had some other issues with blueman, was still getting connection error messages  from time to time, and that was annoying.

Probably the issues I still had with blueman is related to the fact that there are many things I modified in my Ubuntu. Anyone that have a default and clean Ubuntu install, should not have any issue.

kuraga, 2022-12-31:
What helped me to connect [Sony WH-1000X4 headphones: put `bluez_monitor.properties["bluez5.codecs"] = "[ sbc sbc_xq aac ldac aptx aptx_hd ]"` in `/etc/wireplumber/bluetooth.lua.d/51-bt-limit-codecs.lua`.

kuraga, 2023-01-09:
What helped me to connect Sony WH-1000X4 headphones: put
```
bluez_monitor.properties["bluez5.codecs"] = "[ sbc sbc_xq aac ldac aptx aptx_hd ]"
```
in `/etc/wireplumber/bluetooth.lua.d/51-bt-limit-codecs.lua` (https://github.com/bluez/bluez/issues/411#issuecomment-1275347117).

P.S. https://github.com/blueman-project/blueman/issues/1938#issuecomment-1356125774 is still relevant.

cschramm, 2024-11-04:
The source is contained in what you quoted: https://github.com/bluez/bluez/issues/411. If there are too many codecs in the list — and by default, it's all that are supported — polling them takes too long for the device, so you have to limit it in some way.