blueman-project/blueman #1940: Keyboard Won't Connect After Disconnect

alexsok-bit, 2022-11-21:
1. I turn on the keyboard for pairing
2. I turn on the search for new devices
3. Connecting a keyboard to a computer
4. Set the trusted flag
5. I turn off the keyboard
6. I turn on the keyboard

expect
7. Keyboard connected

really
7. The keyboard cannot connect. The messages "keyboard connected" and immediately "keyboard disconnected" appear.

```
~$ blueman-applet --loglevel debug
blueman-applet 19.39.10 INFO     PluginManager:87 load_plugin: ['DisconnectItems', 'StandardItems', 'ExitItem', 'PowerManager', 'NMDUNSupport', 'AuthAgent', 'AutoConnect', 'NetUsage', 'StatusIcon', 'StatusNotifierItem', 'RecentConns', 'TransferService', 'DiscvManager', 'Networking', 'SerialManager', 'DBusService', 'ConnectionNotifier', 'NMPANSupport', 'Menu', 'DhcpClient', 'ShowConnected', 'PPPSupport', 'GameControllerWakelock', 'KillSwitch']
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 19.39.10 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '0C:DD:24:A4:39:E9', 'AddressType': 'public', 'Name': 'Ooghaejughaejoo0', 'Alias': 'Ooghaejughaejoo0', 'Class': 8126724, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 19.39.10 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 19.39.10 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 19.39.10 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 19.39.11 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 19.39.11 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 19.39.11 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 19.39.11 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 19.39.11 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 19.39.11 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 19.39.11 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 19.39.11 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 19.39.11 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 19.39.11 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 19.39.11 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 19.39.11 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.55
blueman-applet 19.39.11 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.6
blueman-applet 19.39.11 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Address': 'DC:2C:26:29:DD:3D', 'AddressType': 'public', 'Name': 'Keychron K3', 'Alias': 'Keychron K3', 'Class': 1344, 'Icon': 'input-keyboard', 'Paired': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001000-0000-1000-8000-00805f9b34fb', '00001124-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v05ACp024Fd0106', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'WakeAllowed': True}
blueman-applet 19.39.11 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 19.39.11 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 19.39.11 INFO     Networking:71 set_nap   : set nap False
blueman-applet 19.39.11 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 19.39.11 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Pairable': True}
blueman-applet 19.39.11 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Pairable True
blueman-applet 19.39.11 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 19.39.12 INFO     PowerManager:140 request_power_state: Requesting True
blueman-applet 19.39.12 INFO     KillSwitch:138 on_power_state_change_requested: True
blueman-applet 19.39.12 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 19.39.12 INFO     PowerManager:125 check     : callbacks done
blueman-applet 19.39.12 INFO     PowerManager:96 set_adapter_state: True
blueman-applet 19.39.12 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True

# turning on the keyboard

blueman-applet 19.39.27 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Connected': True}
blueman-applet 19.39.27 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Connected': False}
blueman-applet 19.39.30 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Connected': True}
blueman-applet 19.39.30 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Connected': False}
blueman-applet 19.39.32 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Connected': True}
blueman-applet 19.39.32 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_DC_2C_26_29_DD_3D {'Connected': False}
[... 39 lines cut ...]
#AutoConnectTimeout = 60

#DeviceID = bluetooth:1234:5678:abcd

#ReverseServiceDiscovery = true

#NameResolving = true

#DebugKeys = false

#ControllerMode = dual

#MultiProfile = off

FastConnectable = true

[Policy]
#ReconnectUUIDs=00001112-0000-1000-8000-00805f9b34fb, 0000111f-0000-1000-8000-00805f9b34fb, 0000110a-0000-1000-8000-00805f9b34fb

ReconnectAttempts=7

ReconnectIntervals=1, 2, 3

AutoEnable=true
```

infirit, 2022-11-21:
Not sure what's going on. I see a device paired but then BlueZ removes the object a bit later as if that never happened. Does the same happen if you use `bluetoothctl` to connect?

alexsok-bit, 2022-11-23:
same way with `bluetoothctl`
```
[CHG] Device DC:2C:26:29:DD:3D RSSI: -71
[CHG] Device DC:2C:26:29:DD:3D Class: 0x00000540
[CHG] Device DC:2C:26:29:DD:3D Icon: input-keyboard
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
[CHG] Device DC:2C:26:29:DD:3D Connected: yes
[CHG] Device DC:2C:26:29:DD:3D Connected: no
```
All works on Linux Mint 20.3

cschramm, 2022-11-23:
@infirit

> I see a device paired

:thinking: Where?

Looks like some state issue to me. Something like `rm -R /var/lib/bluetooth/*/DC:2C:26:29:DD:3D /var/lib/bluetooth/*/cache/DC:2C:26:29:DD:3D` while `bluetooth.service` is stopped could fix it ... or maybe reloading the adapter's driver.

infirit, 2022-11-23:
> @infirit
> 
> > I see a device paired
> 
> thinking Where?

It's in the screenshot. Not sure why the log and screenshot are different.

MarkusVolk, 2022-11-24:
I also had this issue using blueman 2.34 and bluez 5.65. One of the latest commits fixed it for me. My keyboard can now reconnect.
Eventually related to this one ?
https://github.com/blueman-project/blueman/commit/decb8e52a15ee6ef9e2cc1d3c3903da43e2651f3

Can you try with blueman 2.35?

alexsok-bit, 2022-11-28:
> If doing like this:
> Turn on the keyboard for pairing
> Turn on the search for new devices
> Right-click on my keyboard
> Select trust
> Select pair
> Select connect

Confirm. After following the steps above, everything worked.

closed 2023-02-05 by github-actions[bot]