blueman-project/blueman #1953: Same issues as most of the users. with Linux mint 20.2 blueman cannot find any devices

Spectre1980, 2022-11-29:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Spectre1980, 2022-11-29:
System:
  Kernel: 5.15.0-53-generic x86_64 bits: 64 compiler: gcc v: 11.2.0 Desktop: Cinnamon 5.4.12
    tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Gigabyte product: Z370P D3 v: N/A serial: <superuser required>
  Mobo: Gigabyte model: Z370P D3-CF v: x.x serial: <superuser required>
    UEFI-[Legacy]: American Megatrends v: F1 date: 10/05/2017
CPU:
  Info: 6-core model: Intel Core i7-8700K bits: 64 type: MT MCP arch: Coffee Lake rev: A cache:
    L1: 384 KiB L2: 1.5 MiB L3: 12 MiB
  Speed (MHz): avg: 1572 high: 2000 min/max: 800/4700 cores: 1: 1854 2: 2000 3: 1100 4: 1100
    5: 1103 6: 1781 7: 1900 8: 1900 9: 1762 10: 1000 11: 1998 12: 1368 bogomips: 88796
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel CoffeeLake-S GT2 [UHD Graphics 630] vendor: Gigabyte CometLake-S driver: i915
    v: kernel ports: active: HDMI-A-1 empty: none bus-ID: 00:02.0 chip-ID: 8086:3e92
  Device-2: Generalplus GENERAL WEBCAM type: USB driver: snd-usb-audio,uvcvideo bus-ID: 1-7:4
    chip-ID: 1b3f:2247
  Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: modesetting unloaded: fbdev,vesa
    gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96
  Monitor-1: HDMI-1 mapped: HDMI-A-1 model: Lenovo LEN T23i-10 res: 1920x1080 dpi: 96
    diag: 584mm (23")
  OpenGL: renderer: Mesa Intel UHD Graphics 630 (CFL GT2) v: 4.6 Mesa 22.0.5 direct render: Yes
Audio:
  Device-1: Intel 200 Series PCH HD Audio vendor: Gigabyte driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:a2f0
  Device-2: Generalplus GENERAL WEBCAM type: USB driver: snd-usb-audio,uvcvideo bus-ID: 1-7:4
    chip-ID: 1b3f:2247
  Sound Server-1: ALSA v: k5.15.0-53-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Gigabyte driver: r8169
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: e000 bus-ID: 06:00.0 chip-ID: 10ec:8168
  IF: enp6s0 state: down mac: <filter>
  Device-2: 802.11g Adapter [Linksys WUSB54GC v3] WUSB100 v2 RangePlus Wireless Network [Ralink
  RT3070]
    type: USB driver: rt2800usb bus-ID: 1-10:6 chip-ID: 1737:0078
  IF: wlx98fc11c9eadd state: up mac: <filter>
Bluetooth:
  Device-1: Cambridge Silicon Radio Bluetooth Dongle (HCI mode) type: USB driver: btusb v: 0.8
    bus-ID: 1-4:3 chip-ID: 0a12:0001
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 1.2 lmp-v: 2.0
    sub-v: 7a6
Drives:
  Local Storage: total: 558.92 GiB used: 27.61 GiB (4.9%)
  ID-1: /dev/sda vendor: Kingston model: SA400S37120G size: 111.79 GiB speed: 6.0 Gb/s
    serial: <filter>
  ID-2: /dev/sdb vendor: Kingston model: SV300S37A480G size: 447.13 GiB speed: 6.0 Gb/s
    serial: <filter>
Partition:
  ID-1: / size: 106.38 GiB used: 27.35 GiB (25.7%) fs: ext4 dev: /dev/dm-1 mapped: vgmint-root
  ID-2: /boot size: 1.61 GiB used: 270.8 MiB (16.5%) fs: ext4 dev: /dev/sda3
  ID-3: /boot/efi size: 512 MiB used: 5.2 MiB (1.0%) fs: vfat dev: /dev/sda2
Swap:
  ID-1: swap-1 type: partition size: 980 MiB used: 0 KiB (0.0%) priority: -2 dev: /dev/dm-2
    mapped: vgmint-swap_1
Sensors:
  System Temperatures: cpu: 27.8 C mobo: 16.8 C
  Fan Speeds (RPM): N/A
Repos:
  Packages: apt: 2131
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com vanessa main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/vscode.list
    1: deb [arch=amd64,arm64,armhf] http: //packages.microsoft.com/repos/code stable main
Info:
  Processes: 350 Uptime: 1h 10m Memory: 7.64 GiB used: 3.78 GiB (49.5%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.3.0 alt: 11 Client: Cinnamon v: 5.4.12 inxi: 3.3.13

cschramm, 2022-11-29:
What do you mean with "same issues as most of the users"? Sounds like prior knowledge that could be relevant.

Most probably BlueZ just does not find any devices. Did you try `bluetoothctl`?

The `bluetooth.service` logs or kernel messages might have some error or other hint.

Spectre1980, 2022-11-29:
Hello @cschramm, I mean that many friends with different USB dongles have the same issue after the upgrade to Linux mint 21: we all love Blueman but something is wrong. it finds my device but does not pair with it
and does not show it inside the Blueman search window

regards

Looks it 
s@pc:~$ bluetoothctl
Agent registered
[CHG] Controller 00:15:83:3D:0A:57 Discovering: yes
[DEL] Device 25:EA:0E:46:23:B5 25-EA-0E-46-23-B5
[NEW] Device 25:EA:0E:46:23:B5 25-EA-0E-46-23-B5
[CHG] Device 25:EA:0E:46:23:B5 RSSI: -64
[CHG] Device 25:EA:0E:46:23:B5 RSSI: -53
[CHG] Device 25:EA:0E:46:23:B5 RSSI: -65
[CHG] Device 25:EA:0E:46:23:B5 RSSI: -55
[CHG] Device 25:EA:0E:46:23:B5 RSSI is nil
[CHG] Controller 00:15:83:3D:0A:57 Discovering: no
[bluetooth]#

cschramm, 2022-11-29:
If you're missing `25-EA-0E-46-23-B5` in blueman, unchecking `Hide unnamed devices` in the `View` menu probably solves that, but I doubt that it's actually pairable in that state, is it?

Spectre1980, 2022-11-29:
It works.. it was only that checkbox.. it is better you modify by default that you uncheck by default that options or many others will ask you the same question ahah 

thanks and regards

cschramm, 2022-11-29:
So the device was pairable without any further steps? :thinking:

Spectre1980, 2022-11-29:
> So the device was pairable without any further steps? thinking

yes.. I suggest to put that option as default.

regards

closed 2022-11-29 by Spectre1980

cschramm, 2022-11-29:
What kind of device is this and do you happen to have another system like Windows, Android, macOS, or iOS on hand? Do other systems show that unnamed device?

cschramm, 2022-12-04:
Also, in case you're able to replicate, could you please try the following:

- Start with the device removed from the system.
- Make the device discoverable.
- Open blueman-manager.
- Run `bluetoothctl scan on` in a terminal.

Wait for a while (>> 1 min). Does the device show up (in case you have unnamed devices hidden) or get a name (in case you have them displayed)?