blueman-project/blueman #1956: GUI right-click context menu always for the first item

davewood, 2022-11-30:
blueman: blueman/testing,now 2.3.4-1+b1 amd64 [installed]
BlueZ: bluez/testing,now 5.66-1 amd64 [installed,automatic]
Distribution: debian bookworm/sid
Desktop environment: xfce4/testing,testing,now 4.16 all [installed,automatic]

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

 - open GUI `blueman-manager`
 - click `Search`
 - n devices appear
 - right click 3rd device
 - select `info`
 - the info window shows data for the 1st device and all other actions in the context menu also affect the first device

this happens on 2 of my computers independently, both run the same versions though. the other computer runs a different window manager. (sway)

cschramm, 2022-11-30:
Fixed in 2.3.5

closed 2022-11-30 by cschramm

davewood, 2022-11-30:
didnt find this issue before:
related https://github.com/blueman-project/blueman/issues/1905