blueman-project/blueman #1957: There is a transparent gap on the left from menu items

kuraga, 2022-12-01:
blueman: 2.3.5
BlueZ: 5.6.5
Distribution: Calculate Linux (Gentoo)
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

There is a transparent gap on the left from menu items:

[image]

I'm talking not about beauty now, but could it be a symptom to some UI issues?

cschramm, 2022-12-01:
That's most probably not rendered by blueman but xfce4-panel, using blueman-applet's StatusNotifierItem plugin. If you disable that plugin, `blueman-tray` will render it.

kuraga, 2022-12-01:
> If you disable that plugin

Yes, behavior changes. There is no a gap.

Feel free to close.

closed 2022-12-03 by cschramm