blueman-project/blueman #1960: External icons are broken

cschramm, 2022-12-05:
#1893 breaks things where we provide icon names other than `blueman` externally, at least:

- The `blueman-adapters.desktop` file where we reference `blueman-device`.
- The `StatusNotifierItem` implementation where we forward whatever icon name the applet provides, currently at least `blueman-tray`, `blueman-active`, `blueman-disabled` and `-symbolic` counterparts.

_Originally posted by @cschramm in https://github.com/blueman-project/blueman/issues/1893#issuecomment-1333540901_

infirit, 2022-12-05:
I wanted to open an issue for those, thanks.

The desktop file icon I'll have it installee like we used to.

For the others I have a solution but I need to find a good way to scale the icons.

infirit, 2022-12-05:
@cschramm What documentation did you use to implement `com.canonical.dbusmenu`? I am trying to find what `Gvariant` format `icon-data` requires and what I found isn't very helpful (`G_VARIANT_TYPE_VARIANT` isn't helping me much).

cschramm, 2022-12-06:
You seem to be new to libappindicator fun land. :wink:

I'd say the authoritative source for "documentation" is https://bazaar.launchpad.net/~dbusmenu-team/libdbusmenu/trunk.16.10/view/head:/libdbusmenu-glib/dbus-menu.xml. There is no actual specification, everybody just seems to build around libappindicator's behavior. The `GVariant` they set (from `libdbusmenu-gtk/parser.c`'s `update_icon`) is a byte array containing PNG data. One can only guess or check compatibility throughout implementations of StatusNotifierHost.

infirit, 2022-12-06:
> You seem to be new to libappindicator fun land. wink

I was genuinely hoping to never having to touch it :fearful: but alas I broke it so now I have to unbreak it :slightly_frowning_face: .

I have linked a draft PR which sends png byte objects instead of icon names. It works but needs some works still which I will do later this week.

infirit, 2022-12-09:
Closing for now.

closed 2022-12-09 by infirit