blueman-project/blueman #1963: Icons are not installed by "make install"

alxchk, 2022-12-07:
blueman: latest git (dcbfd9dba3bf8d1a9d2323ecc6c86f4f84e13dd7)

```
git clone https://github.com/blueman-project/blueman /tmp/blueman
autoreconf -i
./configure
make DESTDIR=/tmp/test2 install
```

```
> find /tmp/test2 -name "*.svg" -or -name "*.png"
/tmp/test2/usr/local/share/icons/hicolor/scalable/apps/blueman.svg

 > find /tmp/blueman -name "*.svg" -or -name "*.png" | wc -l
116
```

cschramm, 2022-12-07:
Thanks for reporting. That's actually on purpose but causes issues, see #1960.

closed 2022-12-07 by cschramm

infirit, 2022-12-07:
@alxchk Yes. All ui, iconsband pixmaps are bundled into a single gresource file. We only need to register that file and gtk+ will be able to find all the files included in it.