blueman-project/blueman #1965: Bluetooth keyboard AuthenticationFailure

HakonHarnes, 2022-12-08:
blueman: 2.3.beta1.r81.gdcbfd9db-1
BlueZ: 1:5.66.r26.g758161c42-1
Distribution: 5.15.81-1-MANJARO

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I am currently unable to pair my bluetooth keyboard. I have successfully been able to pair and connect to the keyboard previously. The issue seems to be intermittent, as I have gotten it to connect multiple times, but it always breaks after some time.

Specifications: 

- Keyboard: BT60v2 
- Kernel: 5.15.81-1-MANJARO 

Note: The keyboard does not require a pin for pairing. I have also trusted the keyboard `bluetoothctl trust <MAC>` before attempting to pair. 

Output from blueman is: 

```
BT60 Connected
```

```
Pairing failed for: 
BT60 (F%:12:39:30:1D:EB)
```

```
BT60 Disconnected 
```

`bluetoothctl pair <MAC>`:

```
Attempting to pair with F5:12:39:30:1D:EB
[CHG] Device F5:12:39:30:1D:EB Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
```

`dmesg | grep Bluetooth`: 

```
[16882.418167] Bluetooth: hci0: unexpected SMP command 0x0b from f5:12:39:30:1d:be
```

`btmon`:

```
@ MGMT Command: Pair Device (0x0019) plen 8                                                                     {0x0001} [hci0] 2.158407
        LE Address: F5:12:39:30:1D:EB (Static)
        Capability: KeyboardDisplay (0x04)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                                                   #1 [hci0] 2.158426
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 60.000 msec (0x0060)
> HCI Event: Command Complete (0x0e) plen 4                                                                           #2 [hci0] 2.273191
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                       #3 [hci0] 2.273284
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                           #4 [hci0] 2.275192
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 45                                                                             #5 [hci0] 2.298224
[... 130 lines cut ...]
@ MGMT Event: Command Complete (0x0001) plen 10                                                                 {0x0001} [hci0] 2.507651
      Pair Device (0x0019) plen 7
        Status: Authentication Failed (0x05)
        LE Address: F5:12:39:30:1D:EB (Static)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                       #20 [hci0] 2.507657
        Handle: 3585 Address: F5:12:39:30:1D:EB (Static)
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                                                            #21 [hci0] 2.508131
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                               #22 [hci0] 2.509139
        Num handles: 1
        Handle: 3585 Address: F5:12:39:30:1D:EB (Static)
        Count: 1
> ACL Data RX: Handle 3585 flags 0x02 dlen 7                                                                         #23 [hci0] 2.551882
      ATT: Exchange MTU Response (0x03) len 2
        Server RX MTU: 65
> HCI Event: Disconnect Complete (0x05) plen 4                                                                       #24 [hci0] 2.552153
        Status: Success (0x00)
        Handle: 3585 Address: F5:12:39:30:1D:EB (Static)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                               {0x0001} [hci0] 2.552193
        LE Address: F5:12:39:30:1D:EB (Static)
        Reason: Connection terminated by local host (0x02)
```

cschramm, 2022-12-08:
I'm far from an expert for this but I can tell: The device responds to the paring request with "paring failed" and I do not see any reason why.

I think the `unexpected SMP command 0x0b` message is not a problem. BlueZ just does not handle the command. Might be an issue for the device to get no response to it, though. :shrug:

HakonHarnes, 2022-12-08:
> I'm far from an expert for this but I can tell: The device responds to the paring request with "paring failed" and I do not see any reason why.
>
>
>
> I think the `unexpected SMP command 0x0b` message [...]

I'm not sure why the device would respond with paring failed, either. I have tried to clear the Bluetooth profile from it before paring, but it doesn't appear to make a difference. The keyboard works flawlessly on other devices like my phone.

cschramm, 2022-12-08:
In case you're able to make it work again, I'd be curious to see a record, but in any case, the far better addressee for this is the BlueZ community.

infirit, 2022-12-08:
Firstly, that a pretty weird blueman version.

If it's like the keyboard i just got then it requires me to enter a pin on the keyboard. The pin is provided as a notification. Maybe you aren't seeing this notification?

HakonHarnes, 2022-12-08:
> Firstly, that a pretty weird blueman version.
> 
> If it's like the keyboard i just got then it requires me to enter a pin on the keyboard. The pin is provided as a notification. Maybe you aren't seeing this notification?

That's the version returned from `pacman -Q`. I have installed the `blueman-git` package from the AUR. Blueman-manager reports version 2.4-alpha. 

The keyboard does not require a pin code. It pairs fine with other devices.

HakonHarnes, 2023-01-17:
I managed to reconnect to the keyboard by uninstalling Manjaro and installing Arch. Still, I ran into the same issue as described above, but now I'm able to resolve it by removing the device and re-pairing, which did not work before. 

@cschramm, here is the output of `btmon` during a successful pairing process: 

```
@ MGMT Command: Pair Device (0x0019) plen 8                                                                    {0x0001} [hci0] 30.768917
        LE Address: F5:12:39:30:1D:EB (Static)
        Capability: KeyboardDisplay (0x04)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                                                #102 [hci0] 30.768959
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 60.000 msec (0x0060)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #103 [hci0] 30.771012
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                    #104 [hci0] 30.771050
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #105 [hci0] 30.773037
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 45                                                                          #106 [hci0] 30.839065
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Random (0x01)
          Address: F5:12:39:30:1D:EB (Static)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -52 dBm (0xcc)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x13
        05 09 42 54 36 30 03 19 c1 03 02 01 06 05 02 12  ..BT60..........
        18 0f 18                                         ...             
        Name (complete): BT60
        Appearance: Keyboard (0x03c1)
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 2 entries
          Human Interface Device (0x1812)
          Battery Service (0x180f)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                    #107 [hci0] 30.839076
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #108 [hci0] 30.840048
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                 #109 [hci0] 30.840058
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
[... 708 lines cut ...]
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                    #245 [hci0] 37.077044
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #246 [hci0] 37.079017
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                                        #247 [hci0] 37.079026
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                                          #248 [hci0] 37.085015
      Inquiry (0x01|0x0001) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                                 {0x0001} [hci0] 37.085024
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
```

cschramm, 2023-01-17:
Weird. The visible flow is exactly the same up to the point where the device sends "pairing failed" in the first log. In the new one it just sends a proper pairing response instead.

HakonHarnes, 2023-01-17:
> Weird. The visible flow is exactly the same up to the point where the device sends "pairing failed" in the first log. In the new one it just sends a proper pairing response instead.

Seems like a hardware issue, but that doesn't make sense since a fresh install of Arch has (partially) resolved the issue. Also, the keyboard works fine with other devices. 

I do dual boot linux and windows, which I can imagine may cause issues. However, it was not paired in windows when the issue occurred.

HakonHarnes, 2023-01-17:
The BT60v2 PCB uses the nordic nRF52840 bluetooth chip. Does that give any clues? Are there any known bluez/blueman issues with this chip?

cschramm, 2023-01-18:
I guess if you could find a known issue, it'd be rather related to the host's chip, not the keyboard's. Unfortunately those things tend to have a memory as well so that e.g. dual booting can cause problems.

Regarding Manjaro vs. Arch you could have been using different driver and firmware versions, but they should be pretty close, yes, and a fresh install should not really have an effect unless the old system's configuration was fucked up in some way.

HakonHarnes, 2023-01-19:
My laptop has the Intel Corporation Comet Lake PCH CNVi WiFi chip. I couldn't find any related issues. I can live with resolving the issue by removing and re-pairing the device every now and then. I'll just have to cross my fingers and hope that workaround continues to work.

HakonHarnes, 2023-02-07:
>  I'll just have to cross my fingers and hope that workaround continues to work.

Well, it has stopped working again. I had to boot into Windows, and restarted into Linux, and the keyboard worked fine... For about two minutes, before it would not pair anymore. Same issue as before, now the workaround (removing it in Linux, clearing the Bluetooth profile on the keyboard, trusting, and re-pairing) doesn't work. 

I did not connect the keyboard to Windows via Bluetooth, I connected it using the USB-c cable. I have no idea if the issue is related to dual booting windows, because it did not immediately stop working after booting back into Linux, and I have been able to dual boot between windows and Linux for the past three weeks without any issues. 

Re-installing Linux resolved the issue for me earlier, so I tried to restore back to a snapshot in Linux that was working for me earlier, but it did not resolve the issue either. This leads me to believe it might be hardware related somehow, and that dual booting windows and Linux might have something to do with it. 

What do you thnk @cschramm?

cschramm, 2023-02-09:
> I do not quite understand how re-installing Linux would somehow alter the state of the Bluetooth-related hardware to make it work again.

If switching from Manjaro to Arch brought a different driver / firmware version, I could imagine that it helped fixing things either in the local chipset or due to a different behavior.

From the btmon logs, it clearly looks like it's the remote device that's acting differently. I don't know how to get any further. The people over at the BlueZ or Linux might be more helpful.

HakonHarnes, 2023-02-09:
> If switching from Manjaro to Arch brought a different driver / firmware version, I could imagine that it helped fixing things either in the local chipset or due to a different behavior.
>
> From the [...]

Thanks for taking the time to have a look at this issue for me and understanding the logs. 

I actually posted an issue on BlueZ, even before posting here, but I got no replies over there.

closed 2023-04-19 by github-actions[bot]