blueman-project/blueman #1967: Blutooth is always getting enabled after opening lid even though it was deactivated and Autoenable is set to false

JulezJay, 2022-12-11:
blueman: 2.3.4+mint1+vanessa
BlueZ: 5.64-Oubunto1
Distribution: Mint21
Desktop environment: Cinamon

* As the title says my Bluetoothadapter on my HP-Envy x360 13-bd0350ng is activated always when I open the lid. It was deactivated before closing the lid (I very seldom use Bluetooth). I have turned off Autoenable in the Manager of Blueman as well as in the config file under /etc/bluetooth/main.conf as well as with this command in the terminal: "gsettings set org.blueman.plugins.powermanager auto-power-on false". The adapter stays off during reboot but is getting enabled always after opening the lid.

infirit, 2022-12-14:
Start it in the terminal, make sure bluetooth is disabled, close the lid. Then when you resume post the whole blueman-applet log here. See the following on how to provide the log https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman.

cschramm, 2022-12-14:
You can also just terminate blueman (e.g. right-click => Exit on the tray icon) before closing the lid to check if it is even involved.

JulezJay, 2022-12-15:
I just checked that if I quit the Bluetooth-Manager and come back from suspend (open the lid or wake up from sleepmode) Bluetooth is deactivated. So it should be caused by the Manager.  I did as @infirit said and activated the debugging, suspended and woke up. This is the terminal output I got:

blueman-applet --loglevel debug
blueman-applet 15.24.08 INFO     PluginManager:87 load_plugin: ['TransferService', 'DisconnectItems', 'DBusService', 'StatusNotifierItem', 'DhcpClient', 'RecentConns', 'NMDUNSupport', 'ShowConnected', 'Menu', 'AutoConnect', 'NetUsage', 'AuthAgent', 'SerialManager', 'KillSwitch', 'StandardItems', 'GameControllerWakelock', 'DiscvManager', 'StatusIcon', 'NMPANSupport', 'PPPSupport', 'Networking', 'ExitItem', 'PowerManager', 'ConnectionNotifier']
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 15.24.08 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'CC:15:31:3B:3B:A1', 'AddressType': 'public', 'Name': 'julez-ENVY', 'Alias': 'julez-ENVY', 'Class': 0, 'Powered': False, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 15.24.08 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 15.24.08 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 15.24.08 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 15.24.08 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 15.24.08 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 15.24.08 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 15.24.08 INFO     PowerManager:179 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 15.24.08 INFO     PowerManager:182 update_power_state: Signalling False
blueman-applet 15.24.08 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
[... cut ...] 0)>
blueman-applet 15.24.45 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DisconnectItems.DisconnectItems object at 0x7ff93a8b8f10>
blueman-applet 15.24.45 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7ff93a8b8ee0>
blueman-applet 15.24.45 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.TransferService.TransferService object at 0x7ff93a8b8a30>

infirit, 2022-12-15:
I see we get the Powered property set True on resume but the Killswitch is still on. We then proceed to put the adapter in a good state and turn off the Killswitch. If blueman-applet is not running the last step doesn't happen.

Not sure if we are doing the right thing here, maybe we should follow the Killswitch state and power down the adapter...?

JulezJay, 2022-12-15:
@infirit: Thanks for your efforts to help me! 
As I am new to Linux I am not shure if I understood everything. Somehow the powered property is set to TRUE after I wake up from suspend but the KillSwitch is still on and then turned off by Blueman to match the new powered property. Is the problem on "my" side so something that is caused by my Linux system or the settings in powermanager (PowerManager:203 on_adapter_property_changed: adapter powered on while in off state, turning bluetooth on)?

cschramm, 2022-12-15:
It looks like the killswitch gets unblocked on resume. I don't get why it would behave like that. :thinking:

JulezJay, 2023-01-01:
I updated to Mint 21.1 and after some days it seems like the issue is gone. Now I have another minor bug I can live with. After some hours of use and several hibernates the blueman symbol is gone from the taskbar and blueman is shut down.

closed 2023-01-01 by JulezJay

cschramm, 2023-01-01:
No blueman-applet process? You can run it from the terminal (with `--loglevel debug`) to see the termination. Otherwise there could be something e.g. in .xsession-errors.