blueman-project/blueman #1969: Exception when trying to pair JBL headphones

TheBoneJarmer, 2022-12-14:
blueman: 2.3.4
BlueZ: 5.64-0ubuntu1
Distribution: Linux Mint
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I installed LM21 recently and so far I have only issues with Blueman. From pairing issues to even crashes. However until now I had no error logs, but today I tried disabling all plugins (to prevent the app freezing my whole system) and tried to pair my JBL headphones by double-clicking on the icon on the list after it was found. This time I got the following exception message:

```python
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 243, in on_event_clicked
    self.menu.connect_service(row["device"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 144, in connect_service
    self.set_op(device, _("Connecting…"))
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 105, in set_op
    if inst.SelectedDevice == self.SelectedDevice and not (inst.is_popup and not inst.props.visible):
AttributeError: 'ManagerDeviceMenu' object has no attribute 'SelectedDevice'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_blueman-manager.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 243, in on_event_clicked
    self.menu.connect_service(row["device"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 144, in connect_service
    self.set_op(device, _("Connecting…"))
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 105, in set_op
    if inst.SelectedDevice == self.SelectedDevice and not (inst.is_popup and not inst.props.visible):
AttributeError: 'ManagerDeviceMenu' object has no attribute 'SelectedDevice'
```

infirit, 2022-12-14:
This error was fixed in 2.3.5, specifically commit e6d051f52b27758f3f70e0bada5a7b64e3320b2b

TheBoneJarmer, 2022-12-14:
Ah ok, superb. Any idea when that patch will become available in the package feed for Ubuntu/Mint?

cschramm, 2022-12-14:
It's already in the Ubuntu repositories.

TheBoneJarmer, 2022-12-20:
I checked it out but on Linux Mint the latest version if 2.3.4. Not sure where to go from here. Leave the issue open for now. I will likely open up a topic on the LM forum.

cschramm, 2022-12-20:
You should be able to install https://packages.debian.org/bookworm/amd64/blueman/download or https://packages.ubuntu.com/lunar/amd64/blueman/download without problems (both are built for Python 3.10, matching Ubuntu 22.04 and thus Linux Mint 21).

closed 2022-12-31 by cschramm