blueman-project/blueman #1976: Cant connect bluetooth headset anymore (br-connection unknown)

andreo141, 2022-12-23:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

ThigSchuch, 2022-12-26:
I'm having a similar problem.

My headset have the MAC 7C:96:D2...
I try to connect to it, but instead, the manager tries to connect on the neighbor's TV, yes, try to connect on a random device.
If this not occur, some other device shows up with different MAC from my headset and I get this "br-connection unknown"

I'm using a recent installation of Mint 21.1 and blueman v2.3.4.

cschramm, 2022-12-27:
@ThigSchuch: Broken right-click targets have been fixed in 2.3.5. Workarounds include showing unnamed devices via the "View" menu.

closed 2023-01-11 by github-actions[bot]