blueman-project/blueman #1981: Connection Failed:  Protocol not available

abhijithm2001, 2022-12-27:
blueman:
Connection failed while connecting bluetooth headset.
showing error: " **Protocol not available**"
[image]

cschramm, 2022-12-27:
Your unspecified system lacks an audio server with Bluetooth support.

pakair, 2023-01-03:
`sudo apt install pulseaudio-module-bluetooth`, reboot, and try again.

diveloper53, 2023-01-04:
> Your unspecified system lacks an audio server with Bluetooth support.

It's Parrot OS, I have it too

> `sudo apt install pulseaudio-module-bluetooth`, reboot, and try again.

Doesn't help :(

cschramm, 2023-01-04:
Obviously, if e.g. your audio server happens to be PipeWire (I don't know what's default in Parrot), a pulseaudio plugin is of no help. `libspa-0.2-bluetooth` would be the package for the PW module.

closed 2023-03-14 by github-actions[bot]