blueman-project/blueman #1983: Keyboard not found in device list

gt4w4al, 2022-12-28:
blueman: 2.3.4+mint1+vanessa
BlueZ: 5.64-0ubuntu1
Distribution: Mint 21.1
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi,

My keyboard is a Microsoft QSZ-00005. It is not seen in the Blueman manager, nor with the bluetoothctl command. Not seen at all, even if my computer is set to "always visible", and even if I untick the box "hide the unnamed devices". So it's even worse than other reported issues !

I can also said I tried on another PC, which is on Mint 20.3 Cinnamon (so with Blueberry package), and the connection was made straightaway, the keyboard was working properly.
I can also easily see the keyboard with my mobile phone.

Instructions in the "troubleshooting" page are not 100% clear for me, I am a newbie, starting in Linux, and english is not my own language... What can I provide you for debbuging information ?

@cschramm : you said "_most probably you're facing an issue with your firmware, driver or whatever. Check kernel and bluetooth.service logs in that case._"
How can I check and what I have to do to check kernel and bluetooth.service logs ?

gt4w4al, 2022-12-28:
I just tried to run a live session on this computer with Linux Mint 20.3 XFCE : the bluetooth manager is seeing my phone, my earphones, but not the keyboard (which is still visible for my phone).
So is this information mean something about the hardware/driver ?
Is this issue related to blueman or to the software which enable the bluetooth detection ? BlueZ or other ? Should I post elsewhere ?

gt4w4al, 2022-12-28:
```
System:
  Host: MacPingouin Kernel: 5.15.0-56-generic x86_64 bits: 64
    Desktop: Xfce 4.16.0 Distro: Linux Mint 21.1 Vera
Machine:
  Type: Desktop System: Apple product: iMac8,1 v: 1.0
    serial: <superuser required>
  Mobo: Apple model: Mac-F227BEC8 v: PVT serial: <superuser required>
    UEFI: Apple v: IM81.88Z.00C1.B00.0802091538 date: 02/09/08
CPU:
  Info: dual core model: Intel Core2 Duo E8235 bits: 64 type: MCP cache:
    L2: 6 MiB
  Speed (MHz): avg: 797 min/max: 800/2800 cores: 1: 797 2: 797
Graphics:
  Device-1: AMD RV630/M76 [Mobility Radeon HD 2600 XT/2700] driver: radeon
    v: kernel
  Device-2: Apple Built-in iSight type: USB driver: uvcvideo
  Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: ati,radeon
    unloaded: fbdev,modesetting,vesa gpu: radeon resolution: 1920x1200~60Hz
  OpenGL: renderer: AMD RV630 (DRM 2.50.0 / 5.15.0-56-generic LLVM 13.0.1)
    v: 3.3 Mesa 22.0.5
Audio:
  Device-1: Intel 82801H HD Audio driver: snd_hda_intel
  Sound Server-1: ALSA v: k5.15.0-56-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Broadcom BCM4321 802.11a/b/g/n driver: wl
  IF: wls4 state: up mac: 00:23:12:22:17:ca
  Device-2: Marvell 88E8058 PCI-E Gigabit Ethernet driver: sky2
  IF: ens5 state: down mac: 00:23:32:9c:ef:ee
Bluetooth:
  Device-1: Apple Bluetooth HCI type: USB driver: btusb
  Report: hciconfig ID: hci0 state: up address: 00:23:12:43:C5:93 bt-v: 1.2
Drives:
  Local Storage: total: 298.09 GiB used: 23.01 GiB (7.7%)
  ID-1: /dev/sda vendor: Western Digital model: WD3200AAJS-40VWA1
    size: 298.09 GiB
Partition:
  ID-1: / size: 91.11 GiB used: 9.49 GiB (10.4%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 486 MiB used: 5.2 MiB (1.1%) fs: vfat
    dev: /dev/sda1
  ID-3: /home size: 194.71 GiB used: 13.51 GiB (6.9%) fs: ext4
    dev: /dev/sda4
Swap:
  ID-1: swap-1 type: partition size: 5.59 GiB used: 0 KiB (0.0%)
    dev: /dev/sda3
Sensors:
  System Temperatures: cpu: 41.0 C mobo: N/A gpu: radeon temp: 65.0 C
  Fan Speeds (RPM): cpu: 1197
Info:
  Processes: 199 Uptime: 19m Memory: 5.79 GiB used: 1.31 GiB (22.6%)
  Shell: Bash inxi: 3.3.13
```

cschramm, 2022-12-29:
So, you do see the keyboard in pairable state from both another Linux system and your phone, but not on the system in question, neither with blueman 2.3.4 and "hide unnamed devices" disabled or bluetoothctl on Linux Mint 21 nor on Linux Mint 20.3 (blueman 2.1.2, I assume; no "hide unnamed devices" option then). Did you try discovering the device with `bluetoothctl` by issuing `scan on` in it?

Check `dmesg` and `journalctl -u bluetooth.service` for messages around the time where you expect the system to discover the device. You can also skim `dmesg` from the system start and look for messages with `Bluetooth` or `hci`.

gt4w4al, 2022-12-29:
Thanks Christopher for your kind help.
You are right regarding the description of my issue, except when I tried with Mint 20.3 because the bluetooth manager in this distro is Blueberry. So,

- it's working with my 2021 PC under Mint 20.3,
- but not working with my iMac in Mint 21.1 (Blueman) or in Mint 20.3 (Blueberry) live session.

Can you please explain what to do to "_Check dmesg and journalctl -u bluetooth.service_" for a newbie ?

gt4w4al, 2022-12-29:
mac@MacPingouin:~$ `dmesg | grep -i bluetooth`
[    1.778269] usb 3-1.1: Product: Bluetooth USB Host Controller
[   17.717029] Bluetooth: Core ver 2.22
[   17.717055] NET: Registered PF_BLUETOOTH protocol family
[   17.717057] Bluetooth: HCI device and connection manager initialized
[   17.717061] Bluetooth: HCI socket layer initialized
[   17.717064] Bluetooth: L2CAP socket layer initialized
[   17.717069] Bluetooth: SCO socket layer initialized
[   29.043470] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   29.043475] Bluetooth: BNEP filters: protocol multicast
[   29.043481] Bluetooth: BNEP socket layer initialized
[   53.958654] Bluetooth: RFCOMM TTY layer initialized
[   53.958666] Bluetooth: RFCOMM socket layer initialized
[   53.958673] Bluetooth: RFCOMM ver 1.11

gt4w4al, 2022-12-29:
An extract of :
mac@MacPingouin:~$ `journalctl -u bluetooth.service`
-- Boot 063da4bdc4c14809aa60a2fc46802738 --
déc. 28 23:57:06 MacPingouin systemd[1]: Starting Bluetooth service...
déc. 28 23:57:07 MacPingouin bluetoothd[629]: Bluetooth daemon 5.64
déc. 28 23:57:11 MacPingouin systemd[1]: Started Bluetooth service.
déc. 28 23:57:11 MacPingouin bluetoothd[629]: Starting SDP server
déc. 28 23:57:11 MacPingouin bluetoothd[629]: Bluetooth management interface 1.21 initialized
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac_hq
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac_sq
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac_mq
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSink/aptx_hd
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_hd
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSink/aptx
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq_453
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq_453
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq_512
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq_512
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq_552
déc. 28 23:57:37 MacPingouin bluetoothd[629]: Endpoint registered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq_552
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Terminating
déc. 28 23:58:20 MacPingouin systemd[1]: Stopping Bluetooth service...
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac_hq
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac_sq
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac_mq
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/aptx_hd
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_hd
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/aptx
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq_453
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq_453
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq_512
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq_512
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq_552
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq_552
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Stopping SDP server
déc. 28 23:58:20 MacPingouin bluetoothd[629]: Exit
déc. 28 23:58:20 MacPingouin systemd[1]: bluetooth.service: Deactivated successfully.
déc. 28 23:58:20 MacPingouin systemd[1]: Stopped Bluetooth service.
-- Boot 78790da9215248c58f9e5f9f2c68c08d --

cschramm, 2022-12-29:
Ok, I don't know if and at which time you expected the device to show up but if that's all there's nothing special.

> Did you try discovering the device with `bluetoothctl` by issuing `scan on` in it?

Background: blueman-manager has a 60 seconds scan timeout which at least showed to be a problem for one specific device (keyboard) in combination with the "hide unnamed devices" option. Scanning in bluetoothctl does not time out and the device might show up late or something.

Other chances you have is find the exact device, e.g. use `lsusb` to find the hardware ID (random guess from Internet data: 05AC:820F) and see if you find any discussion on it. Most references I find for my guess are about WiFi not Bluetooth, though.

The easy fix to get proper Bluetooth on the device is to get some cheap USB dongle and ignore the integrated thingy. :sweat_smile:

gt4w4al, 2022-12-30:
> Did you try discovering the device with bluetoothctl by issuing scan on in it?
Yes I tried, and waited for some minutes, but still nothing.

Can I catch the hardware ID on another PC which sees it, and then "force" pairing on my computer ?

cschramm, 2022-12-31:
You mean the address? I guess it's possible to hack it into BlueZ' state, but I don't think you'll get anywhere from that.

More things you could do:

Enable bluetoothd debugging and see if anything shows up in `bluetooth.service` for the device then. https://wiki.ubuntu.com/DebuggingBluetooth#Generating_log_files_with_debugging_information.

Check `btmon` to see if anything and what shows up from the device.

gt4w4al, 2023-01-02:
I meant I can gather the MAC address from another computer and ask for pairing this MAC address. But it didn't worked...

gt4w4al, 2023-01-02:
I launched bluetooth ctl scan on in another Terminal, I can see my phone or my other computer bluetooth with `btmon`but still nothing regarding the keyboard...

Interesting thing : it seems the discovery is looping in a 10s enable/loop.Is it expected ? I heard we need time to discover sometime, but we are here far from the >1min you are expecting in the worst case.
```
mac@MacPingouin:~$ sudo btmon
Bluetooth monitor ver 5.64
= Note: Linux version 5.15.0-56-generic (x86_64)                                                               0.187074
= Note: Bluetooth subsystem version 2.22                                                                       0.187078
= New Index: 00:23:12:43:C5:93 (Primary,USB,hci0)                                                       [hci0] 0.187081
= Open Index: 00:23:12:43:C5:93                                                                         [hci0] 0.187083
= Index Info: 00:23:12:43:C5:93 (Broadcom Corporation)                                                  [hci0] 0.187085
@ MGMT Open: bluetoothd (privileged) version 1.21                                                     {0x0001} 0.187092
@ MGMT Command: Start Discovery (0x0023) plen 1                                                {0x0001} [hci0] 4.028534
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                          #1 [hci0] 4.028602
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                            #2 [hci0] 4.204833
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                 {0x0001} [hci0] 4.204981
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2                                                      {0x0001} [hci0] 4.204998
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> HCI Event: Inquiry Complete (0x01) plen 1                                                         #3 [hci0] 14.712904
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 14.713023
        Address type: 0x01
          BR/EDR
        Discovery: Disabled (0x00)
@ MGMT Command: Start Discovery (0x0023) plen 1                                               {0x0001} [hci0] 19.628530
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                         #4 [hci0] 19.628627
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                           #5 [hci0] 19.812790
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                {0x0001} [hci0] 19.812912
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 19.812942
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> HCI Event: Inquiry Complete (0x01) plen 1                                                         #6 [hci0] 30.336900
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 30.337016
        Address type: 0x01
          BR/EDR
        Discovery: Disabled (0x00)
@ MGMT Command: Start Discovery (0x0023) plen 1                                               {0x0001} [hci0] 35.627475
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                         #7 [hci0] 35.627578
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                           #8 [hci0] 35.800867
[... 19 lines cut ...]
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                        #10 [hci0] 51.629105
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                          #11 [hci0] 51.804787
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                {0x0001} [hci0] 51.804935
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 51.804956
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> HCI Event: Inquiry Complete (0x01) plen 1                                                        #12 [hci0] 62.324901
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 62.325015
        Address type: 0x01
          BR/EDR
        Discovery: Disabled (0x00)

```

gt4w4al, 2023-01-02:
mac@MacPingouin:~$ sudo btmon
Bluetooth monitor ver 5.64
= Note: Linux version 5.15.0-56-generic (x86_64)                                                               0.187074
= Note: Bluetooth subsystem version 2.22                                                                       0.187078
= New Index: 00:23:12:43:C5:93 (Primary,USB,hci0)                                                       [hci0] 0.187081
= Open Index: 00:23:12:43:C5:93                                                                         [hci0] 0.187083
= Index Info: 00:23:12:43:C5:93 (Broadcom Corporation)                                                  [hci0] 0.187085
@ MGMT Open: bluetoothd (privileged) version 1.21                                                     {0x0001} 0.187092
@ MGMT Command: Start Discovery (0x0023) plen 1                                                {0x0001} [hci0] 4.028534
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                          #1 [hci0] 4.028602
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                            #2 [hci0] 4.204833
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                 {0x0001} [hci0] 4.204981
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2                                                      {0x0001} [hci0] 4.204998
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> HCI Event: Inquiry Complete (0x01) plen 1                                                         #3 [hci0] 14.712904
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 14.713023
        Address type: 0x01
          BR/EDR
        Discovery: Disabled (0x00)
@ MGMT Command: Start Discovery (0x0023) plen 1                                               {0x0001} [hci0] 19.628530
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                         #4 [hci0] 19.628627
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                           #5 [hci0] 19.812790
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                {0x0001} [hci0] 19.812912
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 19.812942
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> HCI Event: Inquiry Complete (0x01) plen 1                                                         #6 [hci0] 30.336900
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 30.337016
        Address type: 0x01
          BR/EDR
        Discovery: Disabled (0x00)
@ MGMT Command: Start Discovery (0x0023) plen 1                                               {0x0001} [hci0] 35.627475
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                         #7 [hci0] 35.627578
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                           #8 [hci0] 35.800867
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                {0x0001} [hci0] 35.801013
      Start Discovery (0x0023) plen 1
[... 13 lines cut ...]
@ MGMT Command: Start Discovery (0x0023) plen 1                                               {0x0001} [hci0] 51.629010
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                        #10 [hci0] 51.629105
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                          #11 [hci0] 51.804787
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                {0x0001} [hci0] 51.804935
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 51.804956
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> HCI Event: Inquiry Complete (0x01) plen 1                                                        #12 [hci0] 62.324901
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                     {0x0001} [hci0] 62.325015
        Address type: 0x01
          BR/EDR
        Discovery: Disabled (0x00)

gt4w4al, 2023-01-02:
I set `[bluetooth]# discoverable-timeout 120` but still 10s loop in btmon.
Is it a clue of something ?

cschramm, 2023-01-02:
:facepalm: I just realized the obvious. The keyboard seems to be an LE device and the adapter does not support that.

gt4w4al, 2023-01-02:
I am glad, we can have an explanation of the issue !
How can you guess this information from the data I sent ? I'm interested for future bug solving !

I understand from Google that LE stands for Low Energy (so introduced in 2010 from Bluetooth 4.0).
My iMac8.1 was made in 2008...

So the remaining solutions are :
1. I guess it's silly to open the iMac expecting to substitute the Bluetooth component by a new and recent one...
2. Buy a BT dongle : 15 to 25€ on websites (BT 5.0 or 5.3). The issue is there is only 3 USB slot on this iMac : one for the RF mouse, one for this dongle, just one remaining ! That's why we aimed for a BT keyboard instead of a RF.
3. Send this keyboard back and buy another keyboard (RF or BT but not LE) --> how can we know if the keyboard is LE or not ? This is usually not detailled information. By the way, it makes sense this kind of component to be "LE". They all should be LE, I hope.

Well, I think I will buy a dongle. It seems BT 5.3 is not well taken into account yet, regarding the drivers. So I don't want to buy a 5.3 which will not be usable on Linux because of the driver. Asus BT500 seems compatible.
How can I switch off the BT integrated to the iMac to run only the BT dongle ?
Are the drivers expected to be installed automatically (if not already done) when plugging the dongle ?

Thanks a lot Christopher for solving this issue. It seems obvious now, but I believe I would wander a moment before understanding this. Your help is really appreciated.

infirit, 2023-01-03:
I have an Asus bt500, it's a decent dongle.

cschramm, 2023-01-03:
Exactly that, typically the system requirements for the devices specify Bluetooth 4.0 if it's LE (only), plus your system is just too old to have that built in and I suppose

    Bluetooth:
      Device-1: Apple Bluetooth HCI type: USB driver: btusb
      Report: hciconfig ID: hci0 state: up address: 00:23:12:43:C5:93 bt-v: 1.2

means that it follows 1.2, although from the age I'd have expected at least 2.0. :shrug:

LE devices not showing up on non-LE hardware makes perfect sense to explain the "selective" behavior you reported from `btmon`.

> How can I switch off the BT integrated to the iMac to run only the BT dongle ?

The easiest way is to block it with `rfkill` but you do not need to, you can just ignore it. Frontends like blueman allow you to select the adapter that you want to control.

> Are the drivers expected to be installed automatically (if not already done) when plugging the dongle ?

As far as I know Ubuntu and thus probably Linux Mint does not have a notion of automatic driver installation. You have a good chance to have everything on your system already, though. @infirit could probably say more on bt500 support.

infirit, 2023-01-03:
> @infirit could probably say more on bt500 support.

It's in kernel 5.14 so Mint 21.1 should support it. 

If you want something really well supported look for a [CSR8510](https://www.qualcomm.com/products/csr8510) dongle. I have [this](https://www.trust.com/en/product/18187-manga-bluetooth-4-0-adapter) one from Trust. But it is in a lot of dongles, just avoid the €/$4 ones as those are broken. [TP-link](https://www.tp-link.com/us/home-networking/usb-adapter/ub400/) also sell a dongle based on this chip.

gt4w4al, 2023-01-03:
> I have an Asus bt500, it's a decent dongle.

Thanks for the feedback !

gt4w4al, 2023-01-03:
> edit: Also MS mentions on the techspecs that it requires BT4.0 LE

Yes, I saw, after cschramm suggested this issue, but nothing on the Amazon webpage where I bought the device. And as I were thinking bluetooth is bluetooth (and a basic keyboard is not rocket science), I really thought it was just plug and play !

gt4w4al, 2023-01-03:
> It's in kernel 5.14 so Mint 21.1 should support it.

Thanks again.

I read a topic where they said Cambridge Silicon Radio dongles are known to have a lot of issues with Linux but maybe the ones you are talking are compatible if you know them.
Please note that the 3 dongles you suggested are BT 4.0 instead of Asus BT-500 which is BT 5.0. Given I will spend 16€ (a few bucks more than 4.0), I will choose the 5.0 ! Now, I know !
Ordered, should arrive on friday...

Thanks again for your support.

closed 2023-01-04 by infirit