blueman-project/blueman #1984: Not Searching devices and also Not Discoverable to other devices

WaheedAbbas, 2022-12-29:
blueman: 2.3.4
BlueZ: 5.64-0ubuntu1
Distribution: LinuxMint 21.1
Desktop environment: XFCE
Kernel: 5.15.0-56-generic

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

 I have a Ugreen Bluetooth 5.0 Adapter with chip RTL8761BUV. Blueman is able to detect it but does not search any devices and is also not discoverable by other devices. 
 Things I have tried so far are:
 1) Patch Kernal with  { USB_DEVICE(0x2b89, 0x8761), .driver_info = BTUSB_REALTEK }, but that just fully kills it and is not able to detect the controller anymore so I had to revert the changes.
 2) Uncheck "Hide Unnamed devices"
 3) rfkill unblock
 4) Place /lib/firmware/rtl_bt/rtl8761b_fw.bin and /lib/firmware/rtl_bt/rtl8761b_config.bin files in /lib/firmware/rtl_bt/
 5) Tried Ubuntu 22.10; same result
 
 This is the [result](https://gist.github.com/WaheedAbbas/ff51e16320ec368877293fadc36c75e9) of `sudo dmesg | grep -i bluetooth`

cschramm, 2022-12-30:
The firmware should already be in place from the `linux-firmware` package and I'd clearly expect the `btrtl` driver to jump in. Did you try `modprobe`ing it explicitly?

cschramm, 2022-12-30:
> Patch Kernal with  { USB_DEVICE(0x2b89, 0x8761), .driver_info = BTUSB_REALTEK }

Ah yeah. If that's the correct ID, that entry seems necessary as it just recently got added to Linux 6.2-rc1.

WaheedAbbas, 2022-12-31:
I am not sure how to execute the `modprobe` command. I ran `modprobe rtl8761b`, it showed "Module rtl8761b not found in directory /lib/modules/5.15.0-56-generic". But if I patch Kernel, Blueman no longer detects the controller and shows "no default controller available". Running the `list` command doesn't show any controller either. `lsusb` shows "2b89:8761 Realtek Bluetooth Radio", so I believe PID and VID are correct.

infirit, 2022-12-31:
It may be failing to load the firmware, see https://bugs.launchpad.net/ubuntu/+source/linux-firmware/+bug/1968604

cschramm, 2022-12-31:
The module is `btrtl`, but the way forward is patching in the device and checking what happens with `dmesg`. The bug mentioned by @infirit sounds like Ubuntu or Linux in general gets something wrong with UART vs. USB firmware for that chip, so that linking to the `rtl8761b` files (that should be the UART firmware) from the `rtl8761bu` files (that should be the USB firmware) seems to help in some cases. :shrug:

WaheedAbbas, 2023-01-01:
Thank you. That worked. What I did was:
1) Patch Kernel with `{ USB_DEVICE(0x2b89, 0x8761), .driver_info = BTUSB_REALTEK | BTUSB_WIDEBAND_SPEECH },`. It stopped recognizing the adapter as before but I didn't panic.
2) I installed drivers from https://raw.githubusercontent.com/Realtek-OpenSource/android_hardware_realtek/rtk1395/bt/rtkbt/Firmware/BT/rtl8761b_fw -o /lib/firmware/rtl_bt/rtl8761b_fw.bin
3) Ran `cd /usr/lib/firmware/rtl_bt` and `sudo ln -s rtl8761b_fw.bin rtl8761bu_fw.bin`
4) Reboot: Super Important

cschramm, 2023-01-01:
Just for the record: You should not need external drivers and that repository looks out of date. The `linux-firmware` package ships both `rtl8761b` and `rtl8761bu` (and it looks like there are situations where you need to provide its `b` as `bu` for things to work).

closed 2023-01-02 by WaheedAbbas